jurkovic-nikola/OpenLinkHub #3: Not all devices enumerate on latest firmware

RobotRoss, 2024-07-09:
**Describe the bug**
With the latest firmware from Corsair (2.7.450), the iCue Link Hub will fail to recognize daisy-chained fans. Previous firmware version (unsure of exact number) worked fine and enumerated all hardware correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. ```
{
  "listenPort": 27003,
  "listenAddress": "127.0.0.1",
  "cpuSensorChip": "asusec-isa-0000",
  "cpuPackageIdent": "CPU"
}
```
2. 3 (h100i iCue Link RGB, 2x QX RGB fans). Only the h100i and 1x QX RGB fan will enumerate (the 1st one in the chain)
3. Error from stdout.log

**Expected behavior**
All hardware to have enumerated as on previous firmware.

```
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"info","msg":"Failed to log to file, using default stderr","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: [Temperatures] Loading profile: /usr/bin/OpenLinkHub/database/temperatures/empty
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: [Server] Running REST and WebUI on 127.0.0.1:27003. WebUI is accessible via: http://127.0.0.1:27003
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"warning","msg":"No profile found for device. Probably initial start","serial":"BE37B38044681A5AA0CC68731068C703","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"BE37B38044681A5AA0CC68731068C703","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"BE37B38044681A5AA0CC68731068C703","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"BE37B38044681A5AA0CC68731068C703","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"BE37B38044681A5AA0CC68731068C703","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:51 hortonator OpenLinkHub[1275]: {"error":null,"level":"error","location":"/usr/bin/OpenLinkHub/database/profiles/BE37B38044681A5AA0CC68731068C703.json","msg":"Unable to create new device profile","time":"2024-07-09T19:30:51+01:00"}
Jul 09 19:30:52 hortonator OpenLinkHub[1275]: {"device":{"manufacturer":"Corsair","product":"iCUE LINK System Hub","serial":"BE37B38044681A5AA0CC68731068C703","firmware":"2.7.450","aio":false,"devices":{"13":{"channelId":13,"type":7,"deviceId":"0292F91400340C62038D1E0000","name":"H100i","rpm":0,"temperature":0,"description":"AIO","profile":"Normal","rgb":"static","HasSpeed">
```

**Desktop (please complete the following information):**
 - OS: Fedora 40
-  2.7.450
 - Browser Firefox
 - Version 127

**Additional context**
Firmware update was only done due to addition of new part (iCue Link VRM fan)

jurkovic-nikola, 2024-07-09:
Added new device IDs

RobotRoss, 2024-07-09:
Can confirm that the latest commit has resolved the issue, all devices are enumerating now. Fantastic work!

closed 2024-07-09 by RobotRoss