jurkovic-nikola/OpenLinkHub #4: Not all Fans Lighting/Detected

DotNetRob, 2024-08-03:
**Describe the bug**
Have a total of 7 iCue fans + lcd cooler/fan.  2 fans rgb's turn off when the service starts and appear to be unconfigurable.  No lights working on 2 bottom fans on the right side of picture.  Hooked in series with 3 block + single light.  Only first in block and single are lighting up.  Lights all work on boot.  Note: Back fan is not hooked into hub system.

**To Reproduce**
Steps to reproduce the behavior:
1. All stock configs, nothing changed
2. Only added 1 device echo "KERNEL==\"hidraw*\", SUBSYSTEMS==\"usb\", ATTRS{idVendor}==\"1b1c\", ATTRS{idProduct}==\"0c3f\", MODE=\"0666\"" | sudo tee /etc/udev/rules.d/99-corsair-icuelink.rules
3. 3 fans on radiator block, 3 fans grouped, 1 single fan, cpu lcd screen/fan
4. stdout.log below
[stdout.log](https://github.com/user-attachments/files/16480718/stdout.log)

**Expected behavior**
2 more fans in list? not sure since some fans are controlled by more than one item on the list.  

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop:**
 - OS: Linux Mint wilma 22 x86_64
-  Device Firmware:  2.7.450 ???
 - Browser firefox
 - Version 128.0.3

lsusb -d 1b1c:
Bus 001 Device 002: ID 1b1c:0c3f Corsair iCUE LINK System Hub
Bus 001 Device 003: ID 1b1c:0c4e Corsair CORSAIR iCUE LINK AIO LCD Screen Module

Hub Info
1 line from Hub goes to radiator(3 fans) and cpu, other line goes to 3 grouped fans then to single fan on bottom.

jurkovic-nikola, 2024-08-03:
Hello, how to you have AIO connected to fans ? I cant reproduce this. I Have H150i with 3 QX fans on 1 hub port, and second port has 3 RX fans + 3 QX fans chained.

DotNetRob, 2024-08-03:
Thanks for the quick reply, love the app, very happy to finally be able to control my lights in linux.  

- Connection 1 on the hub goes to the 3x CORSAIR RX120 RGB Fans that where preinstalled in the case, connected via iCUE cable, then the stand alone RX120 is connected via iCUE cable to the 3x block on fans.  
- Connection 2 from the hub goes to the 3x QX120 RGB Fans with the LCD screen only.

I am going to try and change up the cabling a bit to see if it change my results.

I also went thru the devices on the screen and labeled which one is controlling which device.  Oldly the bottom single fan seems to be controled by 3 different listings.
[image]

DotNetRob, 2024-08-03:
Ok, got all the light working, after some trials, I ended up with swapping the connectors on the hub and all the lights work.  I do get some weird results when changing rgb on the fans, most of them result in multiple fans changing.  But I am good with that, I can tweak the rgb values and get it to the colors combinations I want.  Very happy.

closed 2024-08-03 by DotNetRob

jurkovic-nikola, 2024-08-10:
@DotNetRob I've found the issue with LED amount for LCD blocks. I'm waiting for a LCD device delivery to test changes. My best guess, couple of days to arrive.

jurkovic-nikola, 2024-08-19:
Device arrived and issue is reproduced. Issue is in total amount of LEDs with LCD cap combination. Normal pump has 20, pump + lcd has 40 LEDs. This is fixed and will be released in 0.1.5, with feature to control basic LCD modes.