jurkovic-nikola/OpenLinkHub #5: Temperature profile controlled by AiO Sensor (fluid temperature)

stillGP, 2024-08-10:
Is it possible to set a temperature profile that would be controlled by the AiO temperature sensor? The scale is from 34 to 50 every 2 degrees, respectively 32,34,36,38,40,42,44,46,48,50. This is a more stable reading and the fans do not go crazy. By the way, I would like to thank you for the work you put in, I can finally control the fan speeds.

jurkovic-nikola, 2024-08-10:
Hello,

I'll look into it. Considering we already have all the data, this is just a switch what temperature to use with what profile.

jurkovic-nikola, 2024-08-11:
This is implemented in 0.1.1 release. Profile will create default values, but you can change them to fit your needs.

stillGP, 2024-08-11:
Thank you very much, works great. There was one small cosmetic error.
[image]

jurkovic-nikola, 2024-08-11:
Package re-uploaded. Should be OK now.

stillGP, 2024-08-11:
Thank you very much. You are a champion.

closed 2024-08-11 by jurkovic-nikola