jurkovic-nikola/OpenLinkHub #6: RGB - does not change the fan backlight

stillGP, 2024-08-11:
The RGB profile changes the backlight only in the pump and there is no backlight in the fans.
My AiO is LCD iCUE H170i ELITE which is detected like this:
Bus 001 Device 005: ID 1b1c:0c1c Corsair CORSAIR iCUE Commander CORE
Bus 001 Device 009: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers
[image]
[image]
[image]

jurkovic-nikola, 2024-08-11:
Is your fan RGB port wired in same port number as PWM wire ? Like Fan 1 PWM goes to hub PWD port 1 and RGB likewise.

stillGP, 2024-08-11:
The fans and pump are connected to COMMANDER CORE as shown in the photo. One wire goes from the pump to the CPU fan on the motherboard (pump speed).
[image]

jurkovic-nikola, 2024-08-11:
Okay, thats good. Can you replace the OpenLinkHub binary with debug one and send me stdout.log after program is started. Debug version attached. 
[debug.zip](https://github.com/user-attachments/files/16573455/debug.zip)

stillGP, 2024-08-11:
Of course. I'm sending the file.
[stdout.log](https://github.com/user-attachments/files/16573471/stdout.log)

jurkovic-nikola, 2024-08-11:
Okay, LED data looks okay by values. Lets run another debug with new file to see why AIO is not detected. Same process as before, replace binary and upload stdout
[debug.zip](https://github.com/user-attachments/files/16573493/debug.zip)

stillGP, 2024-08-11:
[stdout.log](https://github.com/user-attachments/files/16573507/stdout.log)

jurkovic-nikola, 2024-08-11:
Lets see if this is detect AIO and LED channels on the pump. 
[test.zip](https://github.com/user-attachments/files/16573542/test.zip)

stillGP, 2024-08-11:
It detected the name ;)

[stdout.log](https://github.com/user-attachments/files/16573556/stdout.log)
[image]

jurkovic-nikola, 2024-08-11:
Good, packet for pumo / radiator was missing second one. Lets see led data now per channel. 
[debug.zip](https://github.com/user-attachments/files/16573568/debug.zip)

stillGP, 2024-08-11:
Unfortunately, RGB only works on the pump.
[stdout.log](https://github.com/user-attachments/files/16573641/stdout.log)

jurkovic-nikola, 2024-08-11:
Nah, pump LEDs are not detected. You just have a combination of 3 detected fans with 8 LEDs and your pump has 24 LEDs, whats why 'pump' works, when in reality those 24 LEDS that are working are actually fan LEDs combined. Lets try with this version. 
[debug.zip](https://github.com/user-attachments/files/16573700/debug.zip)

stillGP, 2024-08-11:
[stdout.log](https://github.com/user-attachments/files/16573741/stdout.log)

jurkovic-nikola, 2024-08-11:
You're welcome. Thank you too for testing those multiple versions. 

`{"LED Channel":0,"connected":true,"level":"info","msg":"LED debug","numLEDs":24,"time":"2024-08-11T14:53:16+02:00"}
{"LED Channel":1,"connected":true,"level":"info","msg":"LED debug","numLEDs":8,"time":"2024-08-11T14:53:16+02:00"}
{"LED Channel":2,"connected":true,"level":"info","msg":"LED debug","numLEDs":8,"time":"2024-08-11T14:53:16+02:00"}
{"LED Channel":3,"connected":true,"level":"info","msg":"LED debug","numLEDs":8,"time":"2024-08-11T14:53:16+02:00"}`

This is how it should look like, Channel 0 is always pump, rest are Commander Core physical LED ports. I'm gonna release new version without all those debug info inside.

closed 2024-08-11 by jurkovic-nikola