jurkovic-nikola/OpenLinkHub #7: 0.1.4-beta doesn't work

stillGP, 2024-08-15:
It starts, the fans start working faster and after a while it turns off. I can't connect to the settings http://127.0.0.1:27003/

➜  ~ lsusb -d 1b1c:
Bus 001 Device 004: ID 1b1c:0c1c Corsair CORSAIR iCUE Commander CORE
Bus 001 Device 008: ID 1b1c:0c39 Corsair LCD Cap for Elite Capellix coolers

[stdout.log](https://github.com/user-attachments/files/16627403/stdout.log)

jurkovic-nikola, 2024-08-15:
Resolved and package re-published. I've never tested on empty profile.

stillGP, 2024-08-15:
It works now. This new LCD option changes the font of the fluid temperature but does not want to change to others, e.g. CPU, GPU or combined.
[image]

jurkovic-nikola, 2024-08-15:
Did you replace both web/ and static/ folders ? There is JS change

stillGP, 2024-08-15:
Yes, first I deleted everything and reinstalled it.

jurkovic-nikola, 2024-08-15:
Can you view page source on device and click on the bottom of page on overview.js file and refresh it. 
[image]

Maybe its cache related.

stillGP, 2024-08-15:
It works. You are a master. Thank you.
[image]

jurkovic-nikola, 2024-08-15:
Awesome. I'll implement no-cache in http code so this does not happen.

closed 2024-08-15 by jurkovic-nikola