jurkovic-nikola/OpenLinkHub #10: Prometheus metrics

kondas, 2024-08-16:
It would be a very useful feature to be able to monitor the fan speeds with Prometheus, plus the go application stats of OpenLinkHub. I am happy to help with creating a Grafana dashboard for the metrics.

jurkovic-nikola, 2024-08-16:
You already have data exposed via JSON on http://127.0.0.1:27003/api/devices endpoint. I don't think much people run prometheus on their home machines and scrape metrics. You would need multiple exporters in order to do so, from node_exporter to others. Grafana can process JSON format that OpenLinkHub already provides back.

jurkovic-nikola, 2024-08-20:
Its using standard prometheus client lib, so overall app stats are also included, but I didn't paste it here due to size of it. 

Will be pushed out in 0.1.7 version. 

`# HELP openlinkhub Product information
# TYPE openlinkhub gauge
openlinkhub{firmware="2.8.458",product="iCUE LINK System Hub",serial="40027074EFEBF2568288ACE590128B30"} 1
# HELP openlinkhub_speed Current speed (RPM) of devices.
# TYPE openlinkhub_speed gauge
openlinkhub_speed{aio="false",channelId="1",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1353
openlinkhub_speed{aio="false",channelId="14",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1338
openlinkhub_speed{aio="false",channelId="15",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1363
openlinkhub_speed{aio="false",channelId="16",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1352
openlinkhub_speed{aio="false",channelId="2",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1335
openlinkhub_speed{aio="false",channelId="3",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1341
openlinkhub_speed{aio="false",channelId="4",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1449
openlinkhub_speed{aio="false",channelId="5",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1449
openlinkhub_speed{aio="false",channelId="6",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 1344
openlinkhub_speed{aio="true",channelId="13",description="AIO",label="Not Set",name="iCUE LINK H150i LCD",probe="false",profile="Normal",pump="true",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 2416
# HELP openlinkhub_temperature Current temperature of devices.
# TYPE openlinkhub_temperature gauge
openlinkhub_temperature{aio="false",channelId="1",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 28.399999618530273
openlinkhub_temperature{aio="false",channelId="14",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 30.399999618530273
openlinkhub_temperature{aio="false",channelId="15",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 33.20000076293945
openlinkhub_temperature{aio="false",channelId="16",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 32.20000076293945
openlinkhub_temperature{aio="false",channelId="2",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 28.5
openlinkhub_temperature{aio="false",channelId="3",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 28.200000762939453
openlinkhub_temperature{aio="false",channelId="4",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 28.600000381469727
openlinkhub_temperature{aio="false",channelId="5",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 28.899999618530273
openlinkhub_temperature{aio="false",channelId="6",description="Fan",label="Not Set",name="iCUE LINK QX RGB",probe="false",profile="Normal",pump="false",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 29.200000762939453
openlinkhub_temperature{aio="true",channelId="13",description="AIO",label="Not Set",name="iCUE LINK H150i LCD",probe="false",profile="Normal",pump="true",rgb="static",serial="40027074EFEBF2568288ACE590128B30"} 37.29999923706055`

kondas, 2024-08-20:
Hi! It looks promising, thanks for adding the feature. I'm sure many others will find it useful. And yes, many exporters needed indeed, OpenLinkHub will be the 9th scrape target on my system. I'm eager to try it out, and will provide a dashboard ASAP.

closed 2024-08-20 by jurkovic-nikola

jurkovic-nikola, 2024-08-20:
demo.json included so you dont have to bother with regex stuff