jurkovic-nikola/OpenLinkHub #8: Multiple system devices on dashboard

kondas, 2024-08-16:
**Describe the bug**
There are multiple system devices appearing on the dashboard, without any purpose

**Expected behavior**
I would only like to see the devices that are useful, ie. they have temperature property, which is usable (CPU and GPU currently AFAIK)

**Screenshots**
[image]
ots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: GNU/Linux
-  Device Firmware
 - Browser [e.g. chrome, safari]
 - Version 0.1.4-beta

**Additional context**
Hello @jurkovic-nikola ! Thanks for the great effort You put in this software. Currently this is the only working solution to be able to control Corsair Commander Core XT under linux. I have noticed that OpenLinkHub detect many system devices, and shows them on the dashboard without them having any useful purpose. It would be great to have an option to disable / hide some of them.

jurkovic-nikola, 2024-08-16:
Hello,

I'm aware of this. Home page exists only because I had no idea what to put there. Complete UI is currently under redo and it will look much better. 

[image]

closed 2024-08-19 by jurkovic-nikola

kondas, 2024-08-20:
Hi @jurkovic-nikola, i've tried the latest UI. Seems much better, but I would like to ask You to revisit this topic. I would love to have more devices with temperature property from the hwmon subsystem. Since that is the de-facto standard for interfacing with hardware sensors on Linux, it has quite good support for various devices. 

One of my goals is to have the cooling of the drive bays controlled depending on the drive temperatures. All the temperatures are available via hwmon sysfs, utilizing the drivetemp module
```
drivetemp-scsi-3-0
Adapter: SCSI adapter
temp1:        +43.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +28.0°C, highest = +53.0°C)

drivetemp-scsi-1-0
Adapter: SCSI adapter
temp1:        +43.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +43.0°C, highest = +48.0°C)
```

There are also DIMM temperature data, and temp inputs from the MoBo superio chip which would also be nice to have. 

```
jc42-i2c-1-1b
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +38.5°C  (low  =  +0.0°C)                  ALARM (HIGH, CRIT)
                       (high =  +0.0°C, hyst =  +0.0°C)
                       (crit =  +0.0°C, hyst =  +0.0°C)

drivetemp-scsi-34-0
Adapter: SCSI adapter
temp1:        +38.0°C  (low  =  +0.0°C, high = +73.0°C)
                       (crit low =  +0.0°C, crit = +73.0°C)
                       (lowest = +21.0°C, highest = +38.0°C)

nvme-pci-0a00
Adapter: PCI adapter
Composite:    +34.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

drivetemp-scsi-5-0
Adapter: SCSI adapter
temp1:        +43.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +42.0°C, highest = +46.0°C)

nvme-pci-0500
Adapter: PCI adapter
Composite:    +37.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

drivetemp-scsi-4-0
Adapter: SCSI adapter
temp1:        +42.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +28.0°C, highest = +51.0°C)

drivetemp-scsi-2-0
Adapter: SCSI adapter
temp1:        +43.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +28.0°C, highest = +51.0°C)

nct6798-isa-0290
Adapter: ISA adapter
in0:                      360.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      720.00 mV (min =  +0.00 V, max =  +0.00 V)
in6:                      272.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
[... 68 lines cut ...]
Adapter: PCI adapter
SVI2_Core:   938.00 mV
SVI2_SoC:    994.00 mV
Tdie:         +39.8°C  (high = +95.0°C)
Tctl:         +39.8°C
Tccd1:        +41.0°C
SVI2_P_Core:   4.94 W
SVI2_P_SoC:    3.83 W
SVI2_C_Core:   5.27 A
SVI2_C_SoC:    3.83 A

drivetemp-scsi-3-0
Adapter: SCSI adapter
temp1:        +43.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +28.0°C, highest = +53.0°C)

drivetemp-scsi-1-0
Adapter: SCSI adapter
temp1:        +45.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +43.0°C, highest = +48.0°C)

```
Whats Your opinion about having those incorporated?

jurkovic-nikola, 2024-08-20:
I'll see what can be done about drive temps controlling the fans.

jurkovic-nikola, 2024-08-20:
So, 

In addition to NVME drivers, SATA drivers are added also. 

[image]

When creating a new speed profile, you will be able to select Storage Temperature and your storage device. 

[image]

Apply is same as any other speed profile an a device, via device overview. Profile contains 8 temperature ranges.

```go
	profileStorageTemperature = TemperatureProfileData{
		Sensor: 3,
		Profiles: []TemperatureProfile{
			{Id: 1, Min: 0, Max: 30, Mode: 0, Fans: 40, Pump: 70},
			{Id: 2, Min: 30, Max: 35, Mode: 0, Fans: 40, Pump: 70},
			{Id: 3, Min: 35, Max: 40, Mode: 0, Fans: 40, Pump: 70},
			{Id: 4, Min: 45, Max: 50, Mode: 0, Fans: 50, Pump: 70},
			{Id: 5, Min: 50, Max: 55, Mode: 0, Fans: 60, Pump: 70},
			{Id: 6, Min: 55, Max: 60, Mode: 0, Fans: 70, Pump: 70},
			{Id: 7, Min: 60, Max: 65, Mode: 0, Fans: 80, Pump: 70},
			{Id: 8, Min: 65, Max: 70, Mode: 0, Fans: 100, Pump: 70},
		},
	}
```