jurkovic-nikola/OpenLinkHub #13: Docker container does not start

MTrinkiesJr, 2024-09-09:
**Describe the bug**
I built the docker container from the current source code and tried to start it with the default mounted config.json. But it gives me this error:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x58 pc=0x84a2a7]

goroutine 12 [running]:
OpenLinkHub/src/devices/cpro.(*Device).getDevices(0xc0002de0a0)
        /app/OpenLinkHub/src/devices/cpro/cpro.go:872 +0x1007
OpenLinkHub/src/devices/cpro.Init(0x1b1c, 0xc10, {0xc000206040, 0x10})
        /app/OpenLinkHub/src/devices/cpro/cpro.go:198 +0x6d9
OpenLinkHub/src/devices.Init.func8(0x0?, 0x0?, {0xc000206040?, 0x0?})
        /app/OpenLinkHub/src/devices/devices.go:828 +0x1c
created by OpenLinkHub/src/devices.Init in goroutine 1
        /app/OpenLinkHub/src/devices/devices.go:827 +0x588
```

**Steps to reproduce the behavior:**
1. Your configuration values
config.json:

```
{
  "listenPort": 27003,
  "listenAddress": "127.0.0.1",
  "cpuSensorChip": "coretemp",
  "manual": false,
  "frontend": true,
  "metrics": true
}
```

Container start command:
```
sudo docker run -dt --network host --privileged --name openlinkhub -v /home/{myname}/docker/mounts/openlinkhub/config.json:/opt/OpenLinkHub/config.json openlinkhub

```

2. Number of devices
1

**Expected behavior**
The docker container starts

**Desktop**
 - OS: Debian 12 Bookworm (KDE Plasma)
-  Device Firmware: 
   - Kernel: 6.1.0-25-amd64 (64-bit)

jurkovic-nikola, 2024-09-09:
This was caused by empty profile and missing null check if profile exists. Try to pull latest code, should be okay. Tested on Commander Pro and it started without crash.

MTrinkiesJr, 2024-09-09:
It's working, thank you very much

closed 2024-09-09 by MTrinkiesJr