jurkovic-nikola/OpenLinkHub #14: Reset / Restart after system suspend

keis, 2024-09-09:
**Is your feature request related to a problem? Please describe.**
When resuming after suspending my system using `systemctl suspend` OpenLinkHub is not working until I restart the service.

**Describe the solution you'd like**
OpenLinkHub detects it's no longer in control of the device and either internally resets or exits and lets the system service manager restart the service

**Describe alternatives you've considered**
Alternatively a systemd unit that runs "After=suspend.target" that restarts OpenLinkHub could be created

jurkovic-nikola, 2024-09-09:
What controller / hub you're using ?

keis, 2024-09-09:
This is on fedora 40 and using the iCUE LINK system hub. There's a iCUE LINK H150i and 3 iCUE LINK RX RGB connected to it in case that matters.

jurkovic-nikola, 2024-09-09:
Couple of changes on how is device monitored. Unfortunately I'm unable to reproduce this on fedora 40 Live or installed version with multiple hubs and firmware levels. Latest commit may or may not solve this issue.

keis, 2024-09-09:
No, unfortunately that did not fix it.

I'll see if I can pin point with some logging where it gets stuck

jurkovic-nikola, 2024-09-10:
On what firmware level is your hub ?

keis, 2024-09-10:
OpenLinkHub reports the system hub version as 2.8.458

jurkovic-nikola, 2024-09-10:
Try with this binary. Check content of stdout.log file for following:

`{"data":"00 01 01 00 10 00 19 01 00 00","level":"warning","msg":"Device status changed, trying to recover...","serial":"40027074EFEBF2568288ACE590128B30","time":"2024-09-10T22:00:07+02:00"}
{"data":"00 01 01 00 10 00 19 01 00 00","level":"warning","msg":"Device status changed, trying to recover...","serial":"40027074EFEBF2568288ACE590128B30","time":"2024-09-10T22:00:08+02:00"}
{"data":"00 01 01 00 10 00 19 01 00 00","level":"warning","msg":"Device status changed, trying to recover...","serial":"40027074EFEBF2568288ACE590128B30","time":"2024-09-10T22:00:09+02:00"}
{"data":"00 01 01 00 10 00 19 01 00 00","level":"warning","msg":"Device status changed, trying to recover...","serial":"40027074EFEBF2568288ACE590128B30","time":"2024-09-10T22:00:10+02:00"}
{"data":"00 00 01 05 10 00 19 01 00 00","level":"info","msg":"Device recovery completed","serial":"40027074EFEBF2568288ACE590128B30","time":"2024-09-10T22:00:11+02:00"}`

This code should fire up once device wakes up from sleep, in this case after suspend operation. 

[OpenLinkHub.zip](https://github.com/user-attachments/files/16950218/OpenLinkHub.zip)

keis, 2024-09-10:
Thanks. I'll give that a go. I originally built from source, you don't happen to have a branch with that on it?

jurkovic-nikola, 2024-09-11:
Source updated with logging. Compile it and see what data is logged.

keis, 2024-09-11:
After resuming there was no output at all in stdout.log

When restarting with systemctl I get this

```
{"level":"info","msg":"Stopping device...","serial":"1DDB2677E366455D932B87F7C59F1BC1","time":"2024-09-11T21:29:46+02:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/1DDB2677E366455D932B87F7C59F1BC1.json","msg":"Loaded custom user profile","serial":"1DDB2677E366455D932B87F7C59F1BC1","time":"2024-09-11T21:29:46+02:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/1DDB2677E366455D932B87F7C59F1BC1.json","msg":"Loaded custom user profile","serial":"1DDB2677E366455D932B87F7C59F1BC1","time":"2024-09-11T21:29:47+02:00"}
{"device":{"manufacturer":"Corsair","product":"iCUE LINK System Hub","serial":"1DDB2677E366455D932B87F7C59F1BC1","firmware":"2.8.458","aio":false,"devices":{"1":{"channelId":1,"type":15,"deviceId":"010032F2620359A1A30000DF27","name":"iCUE LINK RX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":0,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false},"13":{"channelId":13,"type":1,"deviceId":"01002BB792035999CE000056E5","name":"iCUE LINK QX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":1,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false},"14":{"channelId":14,"type":1,"deviceId":"01002BB792035999CE00009DE0","name":"iCUE LINK QX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":1,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false},"15":{"channelId":15,"type":7,"deviceId":"02626838003DCF6403A57B0000","name":"iCUE LINK H150i","rpm":0,"temperature":0,"description":"AIO","profile":"Normal","rgb":"static","label":"Not Set","portId":1,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":true},"16":{"channelId":16,"type":1,"deviceId":"01002BB792035999CE0000571F","name":"iCUE LINK QX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":1,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false},"2":{"channelId":2,"type":15,"deviceId":"01001800210364A27E00000B7B","name":"iCUE LINK RX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":0,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false},"3":{"channelId":3,"type":15,"deviceId":"01001B002103642E4F00005881","name":"iCUE LINK RX RGB","rpm":0,"temperature":0,"description":"Fan","profile":"Normal","rgb":"static","label":"Not Set","portId":0,"IsTemperatureProbe":false,"HasSpeed":true,"HasTemps":true,"AIO":false}},"userProfiles":{"default":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/1DDB2677E366455D932B87F7C59F1BC1.json","Product":"iCUE LINK System Hub","Serial":"1DDB2677E366455D932B87F7C59F1BC1","LCDMode":0,"Brightness":0,"SpeedProfiles":{"1":"Normal","13":"Normal","14":"Normal","15":"Normal","16":"Normal","2":"Normal","3":"Normal"},"RGBProfiles":{"1":"static","13":"static","14":"static","15":"static","16":"static","2":"static","3":"static"},"Labels":{"1":"Not Set","13":"Not Set","14":"Not Set","15":"Not Set","16":"Not Set","2":"Not Set","3":"Not Set"}}},"DeviceProfile":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/1DDB2677E366455D932B87F7C59F1BC1.json","Product":"iCUE LINK System Hub","Serial":"1DDB2677E366455D932B87F7C59F1BC1","LCDMode":0,"Brightness":0,"SpeedProfiles":{"1":"Normal","13":"Normal","14":"Normal","15":"Normal","16":"Normal","2":"Normal","3":"Normal"},"RGBProfiles":{"1":"static","13":"static","14":"static","15":"static","16":"static","2":"static","3":"static"},"Labels":{"1":"Not Set","13":"Not Set","14":"Not Set","15":"Not Set","16":"Not Set","2":"Not Set","3":"Not Set"}},"OriginalProfile":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/1DDB2677E366455D932B87F7C59F1BC1.json","Product":"iCUE LINK System Hub","Serial":"1DDB2677E366455D932B87F7C59F1BC1","LCDMode":0,"Brightness":0,"SpeedProfiles":{"1":"Normal","13":"Normal","14":"Normal","15":"Normal","16":"Normal","2":"Normal","3":"Normal"},"RGBProfiles":{"1":"static","13":"static","14":"static","15":"static","16":"static","2":"static","3":"static"},"Labels":{"1":"Not Set","13":"Not Set","14":"Not Set","15":"Not Set","16":"Not Set","2":"Not Set","3":"Not Set"}},"Template":"lsh.html","HasLCD":false,"VendorId":6940,"LCDModes":{"0":"Liquid Temperature","1":"Pump Speed","2":"CPU Temperature","3":"GPU Temperature","4":"Combined"},"Brightness":{"0":"RGB Profile","1":"33 %","2":"66 %","3":"100 %"},"PortProtection":{"0":24,"1":122},"GlobalBrightness":0,"IsCritical":false,"FirmwareInternal":[2,8,458]},"level":"info","msg":"Device successfully initialized","time":"2024-09-11T21:29:47+02:00"}
```

jurkovic-nikola, 2024-09-11:
Try with latest commit. This should re-initialize complete device and log is current usb connection is dead.

keis, 2024-09-11:
I don't see it doing anything with that patch. I tried adding some extra logging and I can see some stuff is still going on

```
diff --git a/src/devices/lsh/lsh.go b/src/devices/lsh/lsh.go
index fec192d..13947df 100644
--- a/src/devices/lsh/lsh.go
+++ b/src/devices/lsh/lsh.go
@@ -940,6 +940,7 @@ func (d *Device) setDeviceStatus() {
        if err != nil {
                logger.Log(logger.Fields{"error": err, "serial": d.Serial}).Error("Unable to write to a device")
        }
+    logger.Log(logger.Fields{"mode": mode}).Info("In setDeviceStatus")
        if len(mode) == 0 {
                // USB header connection is terminated, re-initialize device
                logger.Log(logger.Fields{"serial": d.Serial}).Error("Device connection is terminated. Trying to reinitialize")
@@ -1012,6 +1013,7 @@ func (d *Device) getDeviceData() {
                                if _, ok := d.Devices[i]; ok {
                                        rpm := int16(binary.LittleEndian.Uint16(currentSensor[1:3]))
                                        if rpm > 1 {
+                        logger.Log(logger.Fields{"deviceIdx": i, "rpm": rpm}).Info("Updating RPM")
                                                d.Devices[i].Rpm = rpm
                                        }
                                }
@@ -1033,6 +1035,7 @@ func (d *Device) getDeviceData() {
                                        if _, ok := d.Devices[i]; ok {
                                                temp := float32(int16(binary.LittleEndian.Uint16(currentSensor[1:3]))) / 10.0
                                                if temp > 1 {
+                            logger.Log(logger.Fields{"deviceIdx": i, "temp": temp}).Info("Updating temperature")
                                                        d.Devices[i].Temperature = temp
                                                }
                                        }
```

Logs before suspend

```
{"level":"info","mode":"AAABBRAAGQEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAD6AAD3AAAHAQAAAQEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=","msg":"In setDeviceStatus","time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":1,"level":"info","msg":"Updating RPM","rpm":756,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":2,"level":"info","msg":"Updating RPM","rpm":759,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":3,"level":"info","msg":"Updating RPM","rpm":760,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":968,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":963,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":2362,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":966,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating temperature","temp":24.9,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating temperature","temp":24.7,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating temperature","temp":26.3,"time":"2024-09-11T23:20:54+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating temperature","temp":25.6,"time":"2024-09-11T23:20:54+02:00"}
{"level":"info","mode":"AAABBRAAGQEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAD5AAD3AAAHAQAAAQEAAAEAAAEAAAEAAAEAAAEAAAEAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=","msg":"In setDeviceStatus","time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":1,"level":"info","msg":"Updating RPM","rpm":757,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":2,"level":"info","msg":"Updating RPM","rpm":759,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":3,"level":"info","msg":"Updating RPM","rpm":760,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":968,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":963,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":2360,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":967,"time":"2024-09-11T23:20:55+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating temperature","temp":24.9,"time":"2024-09-11T23:20:55+02:00"}
[... cut ...]
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":249,"time":"2024-09-11T23:21:57+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":247,"time":"2024-09-11T23:21:57+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":263,"time":"2024-09-11T23:21:57+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":255,"time":"2024-09-11T23:21:57+02:00"}
```

Those exact logs then repeat indefinitely. It's interesting how the last temperature update seems to get interpreted as rpms after the suspend resume cycle

jurkovic-nikola, 2024-09-11:
Also, try to replace cmdGetDeviceMode command code. 

From `cmdGetDeviceMode = []byte{0x01, 0x08, 0x01}` to `cmdGetDeviceMode = []byte{0x01, 0x12, 0x00}`. 0x01, 0x12, 0x00 is being used by iCUE while waiting for hub to became active.

keis, 2024-09-12:
This is the stdout with those changes

```
{"level":"info","mode":"00 00 01 05 10 00 19 01 00 00","msg":"In setDeviceStatus","time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":1,"level":"info","msg":"Updating RPM","rpm":756,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":2,"level":"info","msg":"Updating RPM","rpm":759,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":3,"level":"info","msg":"Updating RPM","rpm":760,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":967,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":963,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":2334,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":965,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating temperature","temp":23.2,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating temperature","temp":22.8,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating temperature","temp":22.5,"time":"2024-09-12T17:25:18+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating temperature","temp":23.4,"time":"2024-09-12T17:25:18+02:00"}
{"level":"info","mode":"00 00 01 05 10 00 19 01 00 00","msg":"In setDeviceStatus","time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":1,"level":"info","msg":"Updating RPM","rpm":757,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":2,"level":"info","msg":"Updating RPM","rpm":760,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":3,"level":"info","msg":"Updating RPM","rpm":761,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":968,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":963,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":2365,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":965,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating temperature","temp":23.2,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating temperature","temp":22.8,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating temperature","temp":22.5,"time":"2024-09-12T17:25:19+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating temperature","temp":23.4,"time":"2024-09-12T17:25:19+02:00"}
{"level":"info","mode":"00 00 01 05 10 00 19 01 00 00","msg":"In setDeviceStatus","time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":232,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":228,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":225,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":234,"time":"2024-09-12T17:25:37+02:00"}
{"level":"info","mode":"00 00 01 05 10 00 19 01 00 00","msg":"In setDeviceStatus","time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":13,"level":"info","msg":"Updating RPM","rpm":232,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":14,"level":"info","msg":"Updating RPM","rpm":228,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":15,"level":"info","msg":"Updating RPM","rpm":225,"time":"2024-09-12T17:25:37+02:00"}
{"deviceIdx":16,"level":"info","msg":"Updating RPM","rpm":234,"time":"2024-09-12T17:25:37+02:00"}
{"level":"info","mode":"00 00 01 05 10 00 19 01 00 00","msg":"In setDeviceStatus","time":"2024-09-12T17:25:38+02:00"}
```

jurkovic-nikola, 2024-09-12:
Okay, I've plugged 8 hubs and couldn't make it to show 00 00 after resume. It's always 00 01 until hub re-initializes. 

Pull latest code and compile, replace config.json with new one and change `"dbusMonitor": false` to `true`. This will completely remove device monitor via actual hub, and will listen of DBus events on suspend and resume.

keis, 2024-09-15:
Yeah, that did the trick! Thx for all the effort looking into this :)

jurkovic-nikola, 2024-09-15:
Thank you for testing all those changes.

closed 2024-09-15 by jurkovic-nikola