jurkovic-nikola/OpenLinkHub #15: H150i  new id

barecool, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
No control displayed in the dashboard for CORSAIR iCUE Link H150i LCD Liquid CPU Cooler - 360mm AIO - QX120 RGB Fans - 2.1” IPS LCD Screen

**Additional context**
config.json
{
  "listenPort": 8080,
  "listenAddress": "0.0.0.0",
  "cpuSensorChip": "coretemp",
  "manual": false,
  "frontend": true,
  "refreshOnStart": false,
  "metrics": true,
  "dbusMonitor": false
}

stdout.log

{"level":"warning","msg":"Unsupported device detected. Please open a new feature request for your device on OpenLinkHub repository","product":3150,"serial":"A6B5T40101PF57","time":"2024-09-14T14:02:02Z","vendor":6940}
{"error":"Failed to open a device with path '/dev/hidraw4': No such file or directory","level":"error","msg":"Unable to open HID device","productId":3135,"serial":"122D323782E03A5EA24BB40B3215782B","time":"2024-09-14T14:02:02Z","vendorId":6940}
{"level":"warning","msg":"Unsupported device detected. Please open a new feature request for your device on OpenLinkHub repository","product":3150,"serial":"A6B5T40101PF57","time":"2024-09-14T14:12:46Z","vendor":6940}
{"error":"Failed to open a device with path '/dev/hidraw4': No such file or directory","level":"error","msg":"Unable to open HID device","productId":3135,"serial":"122D323782E03A5EA24BB40B3215782B","time":"2024-09-14T14:12:46Z","vendorId":6940}
{"level":"warning","msg":"Unsupported device detected. Please open a new feature request for your device on OpenLinkHub repository","product":3150,"serial":"A6B5T40101PF57","time":"2024-09-14T14:20:37Z","vendor":6940}
{"error":"Failed to open a device with path '/dev/hidraw4': No such file or directory","level":"error","msg":"Unable to open HID device","productId":3135,"serial":"122D323782E03A5EA24BB40B3215782B","time":"2024-09-14T14:20:37Z","vendorId":6940}

jurkovic-nikola, 2024-09-14:
Hello,

What OS are you using ?

barecool, 2024-09-15:
sorry to have bugged you I figured the issue out it was a typo on a device mapping 

but I was able to getting all working on 

- Proxmox 8.2.4
- installed docker and portainer on an LXC 
- with usb passthrough (this is where the typo was 
- it all works good now

basically I found the ids for the CORSAIR iCUE Link H150i LCD Liquid CPU Cooler - 360mm AIO - QX120 RGB Fans - 2.1” IPS LCD Screen

 - Under PVE Host Shell 

```
lsusb 

Bus 001 Device 011: ID 1b1c:0c4e Corsair CORSAIR iCUE LINK AIO LCD Screen Module
Bus 001 Device 009: ID 1b1c:0c3f Corsair iCUE LINK System Hub
```

- Created udev rules for the passthrough

```
nano /etc/udev/rules.d/99-usb-serial.rules

SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c3f" SYMLINK+="ttycorsair0"
KERNEL=="hidraw[0-9]*",MODE="0666"
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c4e" SYMLINK+="ttycorsair1"
KERNEL=="hidraw[0-9]*",MODE="0666"
```

- Saved it then and triggered an update of udevadm

```
udevadm trigger
```

- Checked the mapping 
```
ls -la /dev/ttycorsair*
lrwxrwxrwx 1 root root 7 Sep 14 19:17 /dev/ttycorsair0 -> hidraw4
lrwxrwxrwx 1 root root 7 Sep 14 19:17 /dev/ttycorsair1 -> hidraw9
```

Edit LXC Container configuration for passthrough

```
nano /etc/pve/lxc/container.conf

lxc.mount.entry: /dev/ttycorsair0 dev/hidraw4 none bind,optional,create=file
lxc.mount.entry: /dev/ttycorsair1 dev/hidraw9 none bind,optional,create=file
```

- Started LXC container (with portainer and openhublink)
```
docker run -d --network host --privileged -v /root/config.json:/opt/OpenLinkHub/config.json --restart unless-stopped openlinkhub
```
<details>
 <summary>Stdout.log</summary>
<p>

  ```
  {
    "level": "warning",
    "msg": "DeviceProfile is not set, probably first startup",
    "serial": "122D323782E03A5EA24BB40B3215782B",
    "time": "2024-09-15T00:25:05Z"
  }
  {
    "level": "warning",
[... 220 lines cut ...]
    "level": "info",
    "location": "/opt/OpenLinkHub/database/profiles/122D323782E03A5EA24BB40B3215782B.json",
    "msg": "Loaded custom user profile",
    "serial": "122D323782E03A5EA24BB40B3215782B",
    "time": "2024-09-15T00:35:09Z"
  }
  {
    "level": "info",
    "location": "/opt/OpenLinkHub/database/profiles/122D323782E03A5EA24BB40B3215782B.json",
    "msg": "Loaded custom user profile",
    "serial": "122D323782E03A5EA24BB40B3215782B",
    "time": "2024-09-15T00:35:12Z"
  }
  {
    "level": "info",
    "location": "/opt/OpenLinkHub/database/profiles/122D323782E03A5EA24BB40B3215782B.json",
    "msg": "Loaded custom user profile",
    "serial": "122D323782E03A5EA24BB40B3215782B",
    "time": "2024-09-15T00:35:16Z"
  }
  ```
</p></details>

- And now it all works fine
[image]

closed 2024-09-15 by barecool