jurkovic-nikola/OpenLinkHub #16: After restarting the system, the program does not start

stillGP, 2024-09-15:
After restarting the system, OpenLinkHub does not start. 
It starts only after typing in the terminal
➜ ~ cd OpenLinkHub/
chmod +x install.sh
sudo sh install.sh
I had no problem with earlier versions, I added it to autostart and it always started.
[stdout.log](https://github.com/user-attachments/files/17005373/stdout.log)

jurkovic-nikola, 2024-09-15:
Check output of `sudo systemctl status OpenLinkHub.service` command

stillGP, 2024-09-15:
This is what it looks like
[image]

jurkovic-nikola, 2024-09-15:
What happens when you restart service from terminal ? Is device working okay then or ?

stillGP, 2024-09-15:
After starting it in the console it doesn't work.
[image]
Can you tell me what to do?

jurkovic-nikola, 2024-09-15:
Just run `sudo systemctl restart OpenLinkHub.service`

stillGP, 2024-09-15:
It worked. But why do I have to restart the service after starting the system?

jurkovic-nikola, 2024-09-15:
> It worked. But why do I have to restart the service after starting the system?

You shouldn't, since its not working as expected. Where issue appears ? When you restart PC, Sleep / Resume or when you just shutdown and turn the PC back on ?

stillGP, 2024-09-15:
The problem always occurs after starting the computer. What to do to prevent the problem from occurring?

jurkovic-nikola, 2024-09-15:
Replace binary with this once and shutdown / turn on PC. After that upload content of stdout.log, i have some guesses what could be, just need initial data. Don't restart, shutdown and start PC. 

[OpenLinkHub.zip](https://github.com/user-attachments/files/17005539/OpenLinkHub.zip)

stillGP, 2024-09-15:
It's up and running.
[stdout.log](https://github.com/user-attachments/files/17005566/stdout.log)

jurkovic-nikola, 2024-09-15:
Okay, try with this binary. Delay has been introduced before device initialization to wait for device being ready. 
[OpenLinkHub.zip](https://github.com/user-attachments/files/17005585/OpenLinkHub.zip)

stillGP, 2024-09-15:
Unfortunately, it doesn't work.
[stdout.log](https://github.com/user-attachments/files/17005595/stdout.log)

jurkovic-nikola, 2024-09-15:
System lost permission towards the device
`sudo rm -rf /etc/udev/rules.d/99-corsair*`

Then:
`echo "KERNEL==\"hidraw*\", SUBSYSTEMS==\"usb\", ATTRS{idVendor}==\"1b1c\", ATTRS{idProduct}==\"0c32\", MODE=\"0666\"" | sudo tee /etc/udev/rules.d/99-corsair-cc-64.rules`

`echo "KERNEL==\"hidraw*\", SUBSYSTEMS==\"usb\", ATTRS{idVendor}==\"1b1c\", ATTRS{idProduct}==\"0c1c\", MODE=\"0666\"" | sudo tee /etc/udev/rules.d/99-corsair-cc-96.rules`

`echo "KERNEL==\"hidraw*\", SUBSYSTEMS==\"usb\", ATTRS{idVendor}==\"1b1c\", ATTRS{idProduct}==\"0c39\", MODE=\"0666\"" | sudo tee /etc/udev/rules.d/99-corsair-elite-lcd.rules`

`sudo udevadm control --reload-rules
sudo udevadm trigger`

After that try to shutdown / turn on.

stillGP, 2024-09-15:
Unfortunately, it doesn't work.
[image]

jurkovic-nikola, 2024-09-15:
Whats the output of stdout now ? Last log had permission denied towards the device.

stillGP, 2024-09-15:
Sorry for the delay, I had to go out and couldn't reply. I'm sending the file.
[stdout.log](https://github.com/user-attachments/files/17006143/stdout.log)

jurkovic-nikola, 2024-09-15:
`sudo getfacl /dev/hidraw2` Run this command to see device permissions.

stillGP, 2024-09-15:
➜  ~ sudo getfacl /dev/hidraw2
[sudo] hasło użytkownika gp: 
getfacl: Usunięcie wiodącego '/' ze ścieżek bezwzględnych
# file: dev/hidraw2
# owner: root
# group: root
user::rw-
user:gp:rw-
group::rw-
mask::rw-
other::rw-

jurkovic-nikola, 2024-09-15:
Okay, permissions are fine now since device started. Shutdown and turn on PC back again, but do not run and install scripts, those script will replace binary with old one.

stillGP, 2024-09-15:
After restarting the computer it still won't start. I guess the only option left is to restart from the terminal.;(
sudo systemctl restart OpenLinkHub.service
[stdout.log](https://github.com/user-attachments/files/17006517/stdout.log)

jurkovic-nikola, 2024-09-15:
Can you send me stdout without manually starting it ? Log contains error even for http server binding, like there is double process running

stillGP, 2024-09-15:
I send after restarting the computer without manual start
[stdout.log](https://github.com/user-attachments/files/17006551/stdout.log)

jurkovic-nikola, 2024-09-15:
`sudo rm /etc/systemd/system/OpenLinkHub.service`

Run this and reboot.

stillGP, 2024-09-15:
After entering the above command and restarting the computer, the program started. I also joined discord.
[stdout.log](https://github.com/user-attachments/files/17006604/stdout.log)

jurkovic-nikola, 2024-09-15:
Issue is solved over Discord.

closed 2024-09-15 by jurkovic-nikola