jurkovic-nikola/OpenLinkHub #17: missing cpu device and to be able to name them

iocc-git, 2024-09-19:
**Describe the bug**

Not sure what version I used about 1-2 months ago but I miss to be able
to name them. Because my fans isnt in the right order.

Also the cpu itself is missing now. Maybe it thinks "iCUE LINK H150i"
is the cpu? Speed 0 rpm but change of color-theme changes all fans instead.

**To Reproduce**

installed OpenLinkHub-0.2.1-1.x86_64.rpm, no changes
iCUE LINK System Hub
iCUE LINK H150i
iCUE LINK QX RGB        fan
iCUE LINK QX RGB        fan
iCUE LINK QX RGB        fan
[stdout.log](https://github.com/user-attachments/files/17066271/stdout.log)

**Expected behavior**

See describe the bug.

**Screenshots**

[image]

**Desktop (please complete the following information):**

AlmaLinux 9.4 with 6.10.8 kernel
1.6.336
firefox-115

**Additional context**

Nope, but I'll add a feature report separately.

jurkovic-nikola, 2024-09-20:
Click on Set Set bellow the device name.

jurkovic-nikola, 2024-09-20:
Regarding 0 speed and no temperature, I can't reproduce that with same firmware levels on hub. 
[image]

You can try to set `"refreshOnStart": true` once for devices to be re-enumerated, but temps and speed from the pump should be reported back.

jurkovic-nikola, 2024-09-20:
Also, replace the binary from this zip and run it for a 30-60 seconds. Send me stdout.log file after that. There was packet validation implemented in those releases, so debug data is needed. 
[OpenLinkHub.zip](https://github.com/user-attachments/files/17070411/OpenLinkHub.zip)

iocc-git, 2024-09-23:
Hello.

Thanks, it wasnt obvious that "Not set" is something that you can click on and set.

Replaced OpenLinkHub binary and restarted.
Still 0 speed and no temperature.

changed refreshOnStart to true and restarted.
it clearly refreshed and the cpu unit now have speed and temp.

Here is about 10 mins of debug data.
[stdout.log](https://github.com/user-attachments/files/17102378/stdout.log)

It still hasnt changed the order of the devices.
I would suggest that the order would be:

iCUE LINK System Hub, iCUE LINK H150i CPU, iCUE LINK QX RGB FAN 1, iCUE LINK QX RGB FAN 2, iCUE LINK QX RGB FAN 3

I also think it should be included in the dashboard by default.
Like the row below the Temperature Sensors. See screenshot...

[image]
[image]

jurkovic-nikola, 2024-09-23:
Did you update package to 0.2.2 ? You should have arrows bellow the fan / pump icon to move. Try to clear browser cache

jurkovic-nikola, 2024-09-23:
If you have Discord, join. It will be much faster to explain where would you like to see CPU info.  https://discord.gg/ywtSAerx

iocc-git, 2024-09-23:
No, I didnt know it was a new version available.
But now I have. Yes, it works after clearing browser cache.

Ok, I'll be there quite soon :P
[image]

jurkovic-nikola, 2024-09-23:
Changes to dashboard index will be implemented either in 0.2.3 or 0.2.4.

closed 2024-09-23 by jurkovic-nikola