jurkovic-nikola/OpenLinkHub #19: Support for LX120 RGB Fans/ Donation Link OR BTC address

Dirty-Dan, 2024-09-22:
First off, thank you so much, this software is incredible and I was in the same situation you were in. 

I have LX120 RGB fans that use ICUE link. Support would be greatly appreciated, also, please consider adding a donation link, or BTC/Monero address for donations.

jurkovic-nikola, 2024-09-23:
Hello, LX support was added couple of releases back. Please change `/opt/OpenLinkHub/config.json` debug value to true and restart product. Send me stdout.log file. 

There is no donations, this is free and will remain free.

Dirty-Dan, 2024-09-24:
Please see attatched below. 
[stdout.log](https://github.com/user-attachments/files/17106199/stdout.log)

jurkovic-nikola, 2024-09-24:
Thank you. 

`040000` - Hub returned 4 devices back
`07020000051a30323532303131423030443130443635303334434539303030300000` - H150i AIO
`01000000051a30313030313141433132303335354437313730303030333444330000` QX Fan
`01000000051a30313030323433413632303335363130414130303030343631380000` QX Fan
`01000000051a30313030304431333032303335343238314630303030313645300000` QX Fan

Hub never returned data for LX Fan, which is 02 id, so program sees 0 LX fans. I'll try to reproduce this once LX fans arrive, with your firmware level on hub.

jurkovic-nikola, 2024-09-28:
@Dirty-Dan Tested LX fans on 1.8.369 version and they are not operational. You will need at least 2.X.X firmware. Please use iCUE to update device firmware and LX fans will be visible and operational.

Dirty-Dan, 2024-09-29:
Thank you, I upgraded the firmware, and now the fans light up blue, but do not show in the device dashboard, any ideas? Here is attached stdout.log
[stdout.log](https://github.com/user-attachments/files/17182211/stdout.log)

jurkovic-nikola, 2024-09-30:
Try with this:
1. Disable debug data, data is okay and fans are detected now. `"debug": false`
2. `sudo rm /opt/OpenLinkHub/database/profiles/A2CB88B1CC6A6A5B91F5785CE7080FBD.json`
3. Change config value `"refreshOnStart": false` to `"refreshOnStart": true`
4. `systemctl restart OpenLinkHub.service`
5. Change config value `"refreshOnStart": true` to `"refreshOnStart": false`
6. `systemctl restart OpenLinkHub.service`

Dirty-Dan, 2024-09-30:
That solved it! Thank you!!!

closed 2024-09-30 by Dirty-Dan