jurkovic-nikola/OpenLinkHub #20: H150i ELITE RGB hub not showing up.

awptechnologies, 2024-10-23:
First off what an amazing program. I can now control my pump speed and set pump rgb. 
I also have the lighting node pro and it works perfect i can see all my fans and set the speed and rgb.

The featre request is to add support for the fan hub that comes with the h150. I can control the pump but i cant control the fans that are connected to the 6 port hub that connects to the h150.

jurkovic-nikola, 2024-10-23:
Hello,

Can you send output of lsusb command and how are your fans connected to the hub. Both Commander Core hubs are supported.

awptechnologies, 2024-10-23:
Yes as soon as i get a chance i will post them.

jurkovic-nikola, 2024-10-23:
Awesome, thank you. Also check if your fans are being connected from Fan header 1 on the hub, I've managed to reproduce issue of not being able to change speed if I don't connect fans from port 1. Port 1 is closest to the big pump connector.

awptechnologies, 2024-10-23:
i used packages from aur source so i would think it is compiled. Also I have to restart over and over again to get stuff to show up and when i reboot it doesn't work right away without running systemctl restart. a bunch of funky stuff.

awptechnologies, 2024-10-23:
nevermind not the aur that was a different package. Im gonna uninstall and try to build from source.

jurkovic-nikola, 2024-10-23:
No need, I'll push commit today and release a new version.

awptechnologies, 2024-10-23:
also after reboot and even after systemctl restart the lighting node pro gets stuck on a static color and in the web interface i dont get the check mark when changing the option. i will uninstall and wait till tomorrow to reinstall.

jurkovic-nikola, 2024-10-23:
Hello, can you join to discord ? https://discord.gg/VUUXcEAT it will be faster

jurkovic-nikola, 2024-10-24:
Latest commit contains fix for fan enumeration and it should work. I've tested with H150 Elite Capellix with fans being randomly connected to different hub ports and speed can be adjusted. 

Regarding LN Pro being stuck on static default color, this could be permission issue on actual OpenLinkHub folder where profiles are stored. 

You can check permissions by looking what user is running the process. 

`cat /etc/systemd/system/OpenLinkHub.service` - for Debian based systems
`cat /usr/lib/systemd/system/OpenLinkHub.service` - for RPM based systems

Output should contain something like this:
`[Service]
User=openlinkhub
Group=openlinkhub
`

openlinkhub user/group should be owner of application folder. You can check folder permissions with 
`ls -la /opt/OpenLinkHub`

To change folder permissions issue following command: `sudo chown -R openlinkhub:openlinkhub /opt/OpenLinkHub/`

After that restart service: `sudo systemctl restart OpenLinkHub.service`. In short, user that runs service, needs to have R/W permissions to application folder.

awptechnologies, 2024-10-25:
Still having all sorts of problems. I am on an arch system btw. 

On arch the .service file is under /etc/systemd/system  and the user is running under my name.

Doing the chown seems to fix my ln pro but i still dont see the fans associated with the pump hub.

awptechnologies, 2024-10-25:
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is missing, creating initial one.","time":"2024-10-25T02:52:31-04:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is created.","time":"2024-10-25T02:52:31-04:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-25T02:52:31-04:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T02:52:31-04:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T02:52:31-04:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/1B70020EE4045EAFB3AFE05D061C00F5.json","msg":"Loaded custom user profile","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-25T02:52:31-04:00"}
{"level":"info","msg":"No RGB compatible devices found","time":"2024-10-25T02:52:31-04:00"}
{"device":{"manufacturer":"Corsair","product":"Lighting Node PRO","serial":"1B70020EE4045EAFB3AFE05D061C00F5","firmware":"0.10.4","devices":{},"userProfiles":{"default":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/1B70020EE4045EAFB3AFE05D061C00F5.json","Product":"Lighting Node PRO","Serial":"1B70020EE4045EAFB3AFE05D061C00F5","Brightness":0,"RGBProfiles":{},"Labels":{},"ExternalHubs":{"0":{"PortId":0,"ExternalHubDeviceType":0,"ExternalHubDeviceAmount":0},"1":{"PortId":1,"ExternalHubDeviceType":0,"ExternalHubDeviceAmount":0}}}},"DeviceProfile":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/1B70020EE4045EAFB3AFE05D061C00F5.json","Product":"Lighting Node PRO","Serial":"1B70020EE4045EAFB3AFE05D061C00F5","Brightness":0,"RGBProfiles":{},"Labels":{},"ExternalHubs":{"0":{"PortId":0,"ExternalHubDeviceType":0,"ExternalHubDeviceAmount":0},"1":{"PortId":1,"ExternalHubDeviceType":0,"ExternalHubDeviceAmount":0}}},"ExternalLedDeviceAmount":{"0":"No Device","1":"1 Device","2":"2 Devices","3":"3 Devices","4":"4 Devices","5":"5 Devices","6":"6 Devices"},"ExternalLedDevice":[{"Index":1,"Name":"HD RGB Series Fan","Total":12,"Command":0},{"Index":2,"Name":"LL RGB Series Fan","Total":16,"Command":0},{"Index":3,"Name":"ML PRO RGB Series Fan","Total":4,"Command":0},{"Index":4,"Name":"QL RGB Series Fan","Total":34,"Command":0},{"Index":5,"Name":"8-LED Series Fan","Total":8,"Command":0},{"Index":6,"Name":"SP RGB Series Fan","Total":1,"Command":0}],"Template":"lnpro.html","Brightness":{"0":"RGB Profile","1":"33 %","2":"66 %","3":"100 %"},"HasLCD":false,"CpuTemp":0,"GpuTemp":0},"level":"info","msg":"Device successfully initialized","time":"2024-10-25T02:52:31-04:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T02:52:32-04:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T02:52:32-04:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/a2004010228ab3babe49e3c56091005f.json","msg":"Loaded custom user profile","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T02:52:34-04:00"}
{"device":{"manufacturer":"Corsair","product":"iCUE Commander CORE","serial":"a2004010228ab3babe49e3c56091005f","firmware":"2.11.221","devices":{"0":{"channelId":0,"type":0,"deviceId":"AIO-0","name":"H150i ELITE CAPELLIX","rpm":0,"temperature":0,"temperatureString":"","description":"AIO","profile":"Normal","rgb":"static","label":"Set Label","HasSpeed":true,"HasTemps":true,"IsTemperatureProbe":false}},"userProfiles":{"default":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/a2004010228ab3babe49e3c56091005f.json","Product":"iCUE Commander CORE","Serial":"a2004010228ab3babe49e3c56091005f","LCDMode":0,"LCDRotation":0,"Brightness":0,"RGBProfiles":{"0":"static"},"SpeedProfiles":{"0":"Normal"},"Labels":{"0":"Set Label"}}},"DeviceProfile":{"Active":true,"Path":"/opt/OpenLinkHub/database/profiles/a2004010228ab3babe49e3c56091005f.json","Product":"iCUE Commander CORE","Serial":"a2004010228ab3babe49e3c56091005f","LCDMode":0,"LCDRotation":0,"Brightness":0,"RGBProfiles":{"0":"static"},"SpeedProfiles":{"0":"Normal"},"Labels":{"0":"Set Label"}},"TemperatureProbes":null,"Template":"cc.html","HasLCD":false,"VendorId":6940,"LCDModes":{"0":"Liquid Temperature","1":"Pump Speed","2":"CPU Temperature","3":"GPU Temperature","4":"Combined","6":"CPU / GPU Temp","7":"CPU / GPU Load","8":"CPU / GPU Load/Temp"},"LCDRotations":{"0":"default","1":"90 degrees","2":"180 degrees","3":"270 degrees"},"Brightness":{"0":"RGB Profile","1":"33 %","2":"66 %","3":"100 %"},"CpuTemp":0,"GpuTemp":0},"level":"info","msg":"Device successfully initialized","time":"2024-10-25T02:52:34-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-25T02:52:38-04:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/a2004010228ab3babe49e3c56091005f.json","msg":"Loaded custom user profile","serial":"a2004010228ab3babe49e3c56091005f","time":"2024-10-25T03:12:59-04:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/1B70020EE4045EAFB3AFE05D061C00F5.json","msg":"Loaded custom user profile","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-25T03:13:07-04:00"}
[... cut ...] ":"cc-vertical.html","HasLCD":false,"VendorId":6940,"LCDModes":{"0":"Liquid Temperature","1":"Pump Speed","2":"CPU Temperature","3":"GPU Temperature","4":"Combined","6":"CPU / GPU Temp","7":"CPU / GPU Load","8":"CPU / GPU Load/Temp"},"LCDRotations":{"0":"default","1":"90 degrees","2":"180 degrees","3":"270 degrees"},"Brightness":{"0":"RGB Profile","1":"33 %","2":"66 %","3":"100 %"},"CpuTemp":0,"GpuTemp":0},"level":"info","msg":"Device successfully initialized","time":"2024-10-25T04:00:56-04:00"}

jurkovic-nikola, 2024-10-25:
Send me output of `lsusb` and image of pump. This looks like Commander Core does not return any data for fans at all.

awptechnologies, 2024-10-25:
Im actually using my asus controller for fan speed and im using the commander core for the rgb control. This is for all my fans.

jurkovic-nikola, 2024-10-25:
Got it. Yeah, fans wont show up if pwm is not present on connector, RGB is just appended later on when devices are enumerated. I'll see what can be done just to enumerate and display rgb.

jurkovic-nikola, 2024-10-25:
Grab latest 0.2.7 tar and install it, fans should be visible now.

awptechnologies, 2024-10-26:
do i need to remove old to install new or can i just rerun the install.sh in the new 0.2.7

jurkovic-nikola, 2024-10-26:
You can rerun install.sh, script was also changed to run application under dedicated username.

awptechnologies, 2024-10-26:
is there a reason the user openlink has the uid of 959 and gid of 957? shouldn't they be the same?

jurkovic-nikola, 2024-10-26:
IDs are given back by the system when `useradd` command is executed. 
`uid=998(openlinkhub) gid=996(openlinkhub) groups=996(openlinkhub)` this is on my system.

awptechnologies, 2024-10-26:
Weird it would skip a number or not add same number. It works when it works but on reboot it still has issues starting up reliably every time.

jurkovic-nikola, 2024-10-26:
Try to add `After=multi-user.target` in systemd service after `Description`. This will prevent service starting on boot, and it will start when gui is loaded. There is really no difference in starting on boot or running it from command line except maybe hub does not initialize completely when service is started.

awptechnologies, 2024-10-26:
[Unit]
Description=Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs

[Service]
User=openlinkhub
Group=openlinkhub
WorkingDirectory=/opt/OpenLinkHub
ExecStart=/opt/OpenLinkHub/OpenLinkHub
ExecReload=/bin/kill -s HUP $MAINPID
RestartSec=5

[Install]
WantedBy=multi-user.target

This is what it is like now where do i add it?

jurkovic-nikola, 2024-10-26:
After Description

[Unit]
Description=Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs
After=multi-user.target

awptechnologies, 2024-10-26:
worked on this reboot i will keep you updated. Thank you so much for helping out. Now if only someone would create a better g hub for Logitech devices...

jurkovic-nikola, 2024-10-26:
Thank you for testing and reporting issues, all of this is making product better from each release.

awptechnologies, 2024-10-31:
it seems openrgb is conflicting with openlink some how. I disabled all devices in openrgb except my mouse and keyboard. At first my computer would freeze everytime i opened openrgb but after disabling devices it doesn't freeze but openlink still acts funny.

awptechnologies, 2024-10-31:
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:54:19-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:54:40-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:54:45-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write red color to device","time":"2024-10-31T05:54:45-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:54:50-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write green color to device","time":"2024-10-31T05:54:50-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:54:55-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write blue color to device","time":"2024-10-31T05:54:55-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:00-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:06-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write red color to device","time":"2024-10-31T05:55:06-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:11-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write green color to device","time":"2024-10-31T05:55:11-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:16-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write blue color to device","time":"2024-10-31T05:55:16-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:21-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:26-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write red color to device","time":"2024-10-31T05:55:26-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:31-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write green color to device","time":"2024-10-31T05:55:31-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:36-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write blue color to device","time":"2024-10-31T05:55:36-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:41-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:47-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write red color to device","time":"2024-10-31T05:55:47-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:52-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write green color to device","time":"2024-10-31T05:55:52-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:55:57-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write blue color to device","time":"2024-10-31T05:55:57-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:56:02-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:56:07-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write red color to device","time":"2024-10-31T05:56:07-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write to a device","serial":"1B70020EE4045EAFB3AFE05D061C00F5","time":"2024-10-31T05:56:12-04:00"}
{"error":"Connection timed out","level":"error","msg":"Unable to write green color to device","time":"2024-10-31T05:56:12-04:00"}

jurkovic-nikola, 2024-10-31:
2 programs are trying to write to same device, it wont work. Either use openrgb or this.

awptechnologies, 2024-10-31:
i have to use this for pump control. 

I have found that changing the user to my local user and leaving the group as openlink it works fine. all i did was edit it in the openlink.service file.

awptechnologies, 2024-10-31:
Im using open link for all corsair cooling stuff and open rgb for my peripherals. Just had to disable the corsair stuff in open rgb.

jurkovic-nikola, 2024-10-31:
What peripherals do you have ? 

0.3.0 will bring support for:
- K55 CORE
- K65 Pro Mini
- K70 CORE

awptechnologies, 2024-10-31:
i use Logitech mouse and keyboard G502 for mouse and g213 keyboard. I also found that i can control my motherboard lights with openrgb.

A question i had for you to about the latest release. I see it supports the rgb vengence ram but i don't see mine showing up. Is there a reason?

jurkovic-nikola, 2024-10-31:
Yes, go thru bottom section of README for Memory and see if you have visible i2c sensors for smbus. If you do not see smbus, its not worth messing around with lax resources.

jurkovic-nikola, 2024-11-03:
Closing this, since original issue was solved by making service delay start.

closed 2024-11-03 by jurkovic-nikola