jurkovic-nikola/OpenLinkHub #21: Manually set RGB Fan

Techozoom, 2024-10-24:
Hello,

first of all i really like the Software you createdit is great and works flawlessly for me! I only have one question:

I connected my Grahpics Card with an ARGB to Corsair adapter to an iCue Commander Core. I set it up under the Windows ICue Software so that the GPU RGB is detected as a Corsair QL fan. With that i can control the RGB of my GPU with the iCue Software. Is it possible that i can set a fan manually in your software?

 I hope you understand what i mean, english is not my first language.

Greetings
Jonas Sturm

jurkovic-nikola, 2024-10-24:
Hello,

I understand, what adapter do you have for this ? I cant find anything on Corsair shop related to such adapter. I cant implement this without adapter.

jurkovic-nikola, 2024-10-24:
Thank you for the info. I've managed to order such cable locally, and delivery time is around 15 days, so it will take some time to arrive and some time to implement this. Since this is 3 wire adapter, yes...manual setup is required, since 4th wire is actual detection what the device is. 

I'll leave this open until cable arrives and something is implemented.

Techozoom, 2024-10-24:
Thank you for your work! My GPU is connected via the 4 Pin Adapter not the 3 Pin Adapter. You also have to manually set it with the 4 wire adapter because iCue cant detect it. The difference beetween the Corsair 3 Pin and 4 Pin is:

ARGB:   3 pin = 5V, DATA, GND
Corsair: 3 pin = 5V, DATA, GND
Corsair: 4 pin = 5V, DATA, FEEDBACK, GND

I think with the 4 pin adapter you can only use the last free RGB-port on the Corsair controller, all devices after it wont work because the feedback pin is not connected, but im not to sure about that.

So if you have a Corsair commander core and 2 fans connected to it in the first and second RGB port, you have to connect the 4 pin adapter to the third port and every port after it then doenst work because the feedback pin isnt connected. Thats how i think it works but as i said i am not really sure about that.

The corsair 3 pin ports should work as normal ARGB port as far as i know, because the pin layout is the same.

Greetings
Jonas

jurkovic-nikola, 2024-10-24:
What connector does your GPU have ? Is it ARGB or something else ?

Techozoom, 2024-10-24:
I have an Shappire RX 6950XT Nitro+ Pure, it has an 3 Pin ARGB port.

dp1140a, 2024-10-28:
Thank you for building this.
I have the same setup:

- Commander Core XT
- 4 Corsair SP120 RGB fans
- 2 DeepCool RGB fans from an LS520 AIO cooler
    - Hooked up to the Commander with the adapter.  
    - I previously had these hooked into the ARGB moterboard pin.  It did show RGB but I could not find a way to control them.
    - The Commnader does not recognize these fans so now no RGB

It works great witht he exception of the DeepCool Fans

I would definitely +1 the feature request to set RGB colors in the UI.

PS:  Anyone know of a way to get the Commander to recognize the Deep Cool Fans?

jurkovic-nikola, 2024-10-28:
There will be option to set free port in use. You will need to know LED amount on the fan, due to packet length. But that's easy to figure out how much LEDs fan has. I'll keep you updated once cable arrive.

jurkovic-nikola, 2024-11-03:
ARGB option is implemented in 0.3.0

Techozoom, 2024-11-04:
Hello, i tried it out and it works flawlessly with my Graphics card, thank you very much!
Greetings
Jonas Sturm

closed 2024-11-04 by jurkovic-nikola