jurkovic-nikola/OpenLinkHub #22: Easy way to turn on/off all light

iocc-git, 2024-11-03:
**Is your feature request related to a problem? Please describe.**

Easy way to turn on/off all light. Possible time scheduled aswell.

**Describe the solution you'd like**

I cant sleep with the lights on.

My "iCUE LINK System Hub" miss a 0% for "power" turned off.
My CPU and FANs got "off" mode.

However its will take alot of time to that every night, its
faster to just put a hoodie over it, not entirely but enough.

When I think about it I want a on/off switch but also scheduled
lights off and on. A button for on/off on the dashboard and
one for scheduled?

So 22:30 it switches off the lights and turn them on 07:30
every day. Maybe you want to be able to set diffrent times
on all seven days.

**Describe alternatives you've considered**

...

**Additional context**

🙂

jurkovic-nikola, 2024-11-03:
You already have off RGB mode per device. 

[image]

I'm having hard time understanding "off" mode. Once you turn off PC, Link System Hub is off without any power and there is no way fans are lighting up while PCIE power is off, since PCIE power is powering the lights.

iocc-git, 2024-11-07:
Right, now I noticed off there.
It works but never updates the latest choice. It always says circle.

I never turn off my PC - its also router and server.

jurkovic-nikola, 2024-11-07:
Well, guess scheduler can be implemented. I'll see for 0.3.1 or 0.3.2 version.

iocc-git, 2024-11-10:
Great 😊

jurkovic-nikola, 2024-11-11:
Scheduler implemented in 0.3.1

closed 2024-11-11 by jurkovic-nikola

iocc-git, 2024-11-17:
Thanks, works just fine 🙂

iocc-git, 2024-11-19:
After a while I realized that it has a bug.
It will only turn off if the off time and the on time is on the same "day".
So off 22:35 and on (the next day) 07:00 will not work.

No errors and its correct in scheduler.json.
The logfile doesnt say anything about it what I can see.
Off just doesnt happend.

Here is the logfile and scheduler.json.
No off for 2024-11-19 22:35 and the lights is still on.
[scheduler.json](https://github.com/user-attachments/files/17821968/scheduler.json)
[stdout.log](https://github.com/user-attachments/files/17821971/stdout.log)

jurkovic-nikola, 2024-11-20:
There wont be any logs. I'll fix it today with different schedule code.

iocc-git, 2024-12-03:
It works now but with a twist. It turns off the leds 19:55 and starts them again 19:59 🤔
I have
     "rgbOff": "23:45",
    "rgbOn": "08:30"      ...and are running OpenLinkHub-0.3.6-1.x86_64.

I think it got sad when I deleted all profiles except the one that Im
using. But sometimes it seems to be looking for it without my "added" name.

Whats the right way to do?
[stdout.log](https://github.com/user-attachments/files/17999043/stdout.log)

jurkovic-nikola, 2024-12-03:
Hey, I've put same values as yours, lets see what will happen.

iocc-git, 2024-12-08:
Did you notice anything similar to me?

Now Im running OpenLinkHub-0.4.0.
I have also noticed if I press save it will only say  RGB.... successfully updated if its enabled.
But if the enable isnt checked it will not say anything when I press save.

jurkovic-nikola, 2024-12-08:
Couple of things changed, left it on test machine for 2 days. So far so good, no issues for now. Gonna leave it for couple of days more.