jurkovic-nikola/OpenLinkHub #23: LCD screen flickering

sinail1, 2024-11-06:
**Describe the bug**
After installing .deb, I opened the web controller and after the first change the LCD screen began flickering. All options for LCD screen and rbg seems to be working correctly.

**To Reproduce**
Corsair iCUE link h150i LCD

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.10
 - Browser: chromium
 
[image]

jurkovic-nikola, 2024-11-06:
Hello,

I'm unable to reproduce this on Ubuntu 24.10. What were the default RGB after the installation ?

jurkovic-nikola, 2024-11-06:
Also, include file /opt/OpenLinkHub/stdout.log

jurkovic-nikola, 2024-12-04:
Closing due to no response from the author.

closed 2024-12-04 by jurkovic-nikola