jurkovic-nikola/OpenLinkHub #26: Not detecting all RGBs

ghahramani, 2024-11-15:
**Describe the bug**
So I have the following devices:
1. CORSAIR DOMINATOR TITANIUM 64G
2. CORS iCUE LINK TITAN 360 White (3 RGB [fans] for the Radiator and 1 for the CPU)
3. NZXT H9 Elite White Mid Tower (Comes with 3 case RBG fans)

The app only detects the 3-radiator RGB fan (look at the screenshot)

[image]

**To Reproduce**
I just added the `KERNEL=="hidraw*", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c3f", MODE="0666"` to `/etc/udev/rules.d/99-corsair-0c3f.rules`

and changed the `/opt/OpenLinkHub/config.json` to:
```
{
  "debug": true,
  "listenPort": 27003,
  "listenAddress": "0.0.0.0",
  "cpuSensorChip": "coretemp",
  "manual": false,
  "frontend": true,
  "refreshOnStart": true,
  "metrics": true,
  "dbusMonitor": true,
  "memory": true,
  "memorySmBus": "i2c-0",
  "memoryType": 5,
  "exclude": []
}
```

Here is the result of `sudo i2cdetect -l`

```
i2c-0	smbus     	SMBus I801 adapter at 3000      	SMBus adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-5	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-6	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-7	i2c       	NVIDIA i2c adapter 1 at 2:00.0  	I2C adapter
i2c-8	i2c       	NVIDIA i2c adapter 2 at 2:00.0  	I2C adapter
i2c-9	i2c       	NVIDIA i2c adapter 4 at 2:00.0  	I2C adapter
i2c-10	i2c       	NVIDIA i2c adapter 6 at 2:00.0  	I2C adapter
i2c-11	i2c       	NVIDIA i2c adapter 7 at 2:00.0  	I2C adapter
i2c-12	i2c       	NVIDIA i2c adapter 8 at 2:00.0  	I2C adapter
i2c-13	i2c       	NVIDIA i2c adapter 9 at 2:00.0  	I2C adapter
```

And result of `sudo i2cdetect -y 0`:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- 19 -- 1b -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- 49 -- 4b -- -- -- -- 
50: -- 51 -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

**Expected behavior**
I should be able to control the RGB of my two memory cards (it is on but not detected) and also the pump RGB (it is off completely) [Look at the photo]

[image]

It should look like this (Windows):

[image]

**Desktop (please complete the following information):**
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	24.1.2
Codename:	Xahea

jurkovic-nikola, 2024-11-15:
Do you have Memory tab in top menu on Dashboard ? Send me stdout.log file and set `refreshOnStart` to false, you dont need this one each time.

ghahramani, 2024-11-15:
No, I don't have any memory tab, here is the screenshot. I also set `refreshOnStart` to `false` now as you mentioned
[image]
[image]

[stdout.log](https://github.com/user-attachments/files/17769580/stdout.log)

jurkovic-nikola, 2024-11-15:
I guess you ran `echo 'KERNEL=="i2c-0", MODE="0666"' | sudo tee /etc/udev/rules.d/99-corsair-memory.rules` for the memory ?

ghahramani, 2024-11-15:
yeah I have added that too after I raised this issue with no success

[image]

jurkovic-nikola, 2024-11-15:
Do you compile code or use binary ?

jurkovic-nikola, 2024-11-15:
Also, remove your profile and let new binary create a new one. 
`sudo rm /opt/OpenLinkHub/database/profiles/85ED160971774B50B89243A6887E4D5D.json`

jurkovic-nikola, 2024-11-15:
Regarding memory, replace the binary and set debug:true in config.json. Send me stdout.log file. 
[OpenLinkHub_0.3.2_amd64.tar.gz](https://github.com/user-attachments/files/17770629/OpenLinkHub_0.3.2_amd64.tar.gz)
Lets see whats happening with memory dimms

ghahramani, 2024-11-16:
I build it and then run `install.sh`

I upgraded it now, it shows my pump RGB but still not showing the memories

[image]

jurkovic-nikola, 2024-11-16:
Thank you for the log. This kit type appears to be somehow different from the others. For time being, disable memory in config.json. It will take 5-8 business days to receive the kit. Will let you know once new code is released.

ghahramani, 2024-11-16:
Please let me know if I can be any of help, if you want me to execute a command or something to understand the memory kit better I can do that till you receive your kit

jurkovic-nikola, 2024-11-16:
You could run following command:
`sudo i2cdump -y 0 0x51`
`sudo i2cdump -y 0 0x53`
`sudo i2cdump -y 0 0x19`
`sudo i2cdump -y 0 0x1b`

ghahramani, 2024-11-29:
After an in-depth discussion on Discord, we discovered that Intel might be using a different code or locking mechanism, which prevents memory detection. To address this, @jurkovic-nikola introduced a feature in version 0.3.5 that allows users to manually define the part number to retrieve the necessary information. By defining the part number and unloading the spd5118 module, which was locking interactions with the memory, OpenLink successfully detected it.

Kernel version: 6.11.2-4-MANJARO

```
sudo rmmod spd5118
sudo dmidecode -t memory | grep 'Part'
	Part Number: CMP64GX5M2B6000C30  
	Part Number: CMP64GX5M2B6000C30  
```

Here is the `config.json`
```
{
    "cpuSensorChip": "coretemp",
    "dbusMonitor": true,
    "debug": false,
    "decodeMemorySku": false,
    "exclude": [],
    "frontend": true,
    "listenAddress": "127.0.0.1",
    "listenPort": 27003,
    "manual": false,
    "memory": true,
    "memorySku": "CMP64GX5M2B6000C30",
    "memorySmBus": "i2c-6",
    "memoryType": 5,
    "metrics": true,
    "refreshOnStart": false
}
```

`memorySku` is part number that find from grep command and `decodeMemorySku` should be `false`

I think we can close this ticket

jurkovic-nikola, 2024-11-29:
Thanks for the detailed description!

closed 2024-11-29 by jurkovic-nikola