jurkovic-nikola/OpenLinkHub #28: Feature Request: Display PSU sensors in the device dashboard.

Nytem, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
There is no issue related to this.

**Describe the solution you'd like**
In the dashboard, like temperature sensors, a display for any sensors from the PSU.

**Describe alternatives you've considered**
Installing another hardware monitor that shows all sensors in my PC works fine. It'd just be pretty cool to have everything in one place rather than needing an extra app.

**Additional context**
In my PC, I have the 2023 revision of the [Corsair H1000i PSU](https://www.corsair.com/us/en/p/psu/cp-9020259-na/hx1000i-fully-modular-ultra-low-noise-platinum-atx-1000-watt-pc-power-supply-cp-9020259-na). Aside from being a platinum PSU, one of the other main reasons I went with it was for it's ability to be plugged into the internal USB headers to display power monitors inside of Icue. Since I've switched to Linux, I can only see those sensors as stuff like "corsair-psu/in0" inside of a hardware monitor app.

Would probably be of low importance since this is just extra sensor monitors for probably very few of Corsairs PSU's. I was just suggesting it since it is another Icue product that could be displayed alongside the other corsair products in the dashboard.

On an side note however, this PSU did allow me to control its fan speed before. Now it just stays at 1200RPM, which I think was its maximum speed? Maybe that could be added to return that functionality.

Thank you all for your time and work. Just about all of the corsair parts in my PC would be uncontrollable without OpenLinkHub.

Nytem, 2024-11-22:
oh that's awesome, thanks for the quick response!

jurkovic-nikola, 2024-11-27:
@Nytem send me lsusb output of your PSU. 

`lsusb -d 1b1c:psuId -v`

Nytem, 2024-11-27:
sorry, just got home. I'm putting the command into my terminal but it isn't returning anything?

jurkovic-nikola, 2024-11-28:
PSU implemented in 0.3.4

jurkovic-nikola, 2024-11-28:
Regarding 1200 RPM, this is not the maximum fan speed. You need to check dust filters and if they are clean. Default device fan behavior is set to 0 RPM mode, where fan kicks in when temperature is reached. You're either in hot environment or something is blocked the natural flow of air.

Nytem, 2024-11-29:
Updated OpenLinkHub, and this is interesting lol
[image]

Nytem, 2024-11-29:
oh and I realized I was typing part of the lsusb command in wrong, my bad for doing it a bit before going to sleep lol..

Bus 001 Device 015: ID 1b1c:1c1e Corsair HX1000i Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1c1e 
  bcdDevice            1.00
  iManufacturer           1 CORSAIR
  iProduct                2 HX1000i Power Supply
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

jurkovic-nikola, 2024-11-29:
Interesting, this is how it should look like. 
[image]

jurkovic-nikola, 2024-11-29:
Managed to reproduce this, this is caused by using other software to monitor this device, it throws number out. Only once software should query the device in any given time.