jurkovic-nikola/OpenLinkHub #31: Does this support fan speed with XG3?

chrispouliot, 2024-12-03:
This project looks really great, and I was wondering if it supported the XG3 water cooling block fan control if connected to the icue hub? I don’t see the XG3 listed under the icue devices but it has a fan which is controlled by icue link as part of it. 

Thanks!

jurkovic-nikola, 2024-12-03:
Hello, thank you. This device is listed as `iCUE LINK XG3 HYBRID`. If you have this device and you can't see it on the dashboard, enable debug mode in config.json file in /opt/OpenLinkHub, restart service and send me stdout.log.

chrispouliot, 2024-12-06:
Amazing, sorry I missed that - that name is really obvious. Thanks! I don’t currently have it but wanted to make sure it worked before getting a proprietary item like this.

closed 2024-12-06 by jurkovic-nikola