jurkovic-nikola/OpenLinkHub #32: Hydro H100i RGB Platinum is missing

domacikolaci, 2024-12-03:
**Describe the bug**
I own the AIO model Corsair Hydro H100i RGB Platinum SE 240mm. I don't see it in listed supported models nor does it show up after installing this.

**To Reproduce**
Steps to reproduce the behavior:
Not in supported list or after installing this software on a fresh install.

**Expected behavior**
Device be supported and shown in the dashboard for configuration.

**Screenshots**
NA

**Desktop (please complete the following information):**
 - OS: Fedora 41 KDE spin
-  Device Firmware: Unknown
 - Browser: Firefox
 - Version: Version of what?
 
**Additional context**
NA

jurkovic-nikola, 2024-12-03:
Hello,

Send verbose output of lsusb for this device. 

`lsusb -d 1b1c:id-id-of-your-cooler -v`

domacikolaci, 2024-12-03:
Bus 001 Device 006: ID 1b1c:0c19 Corsair H100i RGB SE

jurkovic-nikola, 2024-12-03:
Include -v at the end of your command for verbose output.

domacikolaci, 2024-12-03:
Apologies, I typod the command and didnt notice :)

Bus 001 Device 006: ID 1b1c:0c19 Corsair H100i RGB SE
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c19 H100i RGB SE
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H100i RGB SE
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered

jurkovic-nikola, 2024-12-04:
Device is implemented in 0.3.7

closed 2024-12-05 by jurkovic-nikola