jurkovic-nikola/OpenLinkHub #33: Archlinux AUR Support

ciphorg, 2024-12-06:
### Archlinux Distro Packaging Support

I have created  _Archlinux AUR_ community packages for _OpenLinkHub_. This is my attempt to properly inform you that I enjoy this tool quite a bit, and that wanted to ease the installation for archlinux users (and myself) as best as possible. I have maintained packages before, and am hoping to keep these up to date alongside your releases. 

If you are all OK with this, I will continue to maintain these on the AUR for others to enjoy. My contact info is located in the respective AUR PKGBUILD, or you may reach out to me here.

### AUR Packages
**Package Links**
- [openlinkhub : Latest Source](https://aur.archlinux.org/packages/openlinkhub)
- [openlinkhub-bin : Latest Binary Release](https://aur.archlinux.org/packages/openlinkhub-bin)
- [openlinkhub-git - Current Git Main](https://aur.archlinux.org/packages/openlinkhub-git)

### AUR Package Layout

I've split this into three packages, for now, to support compiling current release, installing the current release binary, or installing from the latest commit on main (for testing). The AUR packaging attempts to mirror `install.sh` & `upgrade.sh` scripts as closely as possible.

- PKGBUILD's are centered on using `systemd` sysusers, tmpfiles, and service-units. 
- Leverage pacman's install/upgrade hooks to handle service restart
- udev device discovery is parsed out into the PKGBUILD

### Background

My machine currently has an _iCUE LINK H150i_, and primarily runs Arch (Kernel latest, lts, and hardened). This allows me to maintain and test packaging and deploying in an Arch-like manner pretty continuously. Just like you I was miffed at the lack of iCUE support on Linux. After coming across your post on the Corsair forums, I've been following this project quite adamantly. 

I hope to submit some PR's related to the classic "timewarp" fan/rgb mode when I get a chance.

jurkovic-nikola, 2024-12-06:
Hello,

I'm fine with this as long as it uses original repository for pulling and building, I would just like to change the way how udev is handled. install.sh script is built for people who want to play with it, while RPM and DEB packages have post installation process to generate udev files for all the devices this program support. 

This is primarily made to avoid user having to reinstall the program, if for an example AIO is upgraded to something new. 

I'll push JSON file with all the devices info in next build, so you can parse it and generate udev rules based on that. 

Feel free to join to Discord on https://discord.gg/2NUN43U4 for more discussion about it.

closed 2024-12-06 by jurkovic-nikola

ciphorg, 2024-12-06:
Thanks for replying,

The AUR packages will absolutely only refer to this project. I have no issue with switching to parsing for and generating the udev rules. 

I have joined discord.