jurkovic-nikola/OpenLinkHub #34: Few bugs + suggestions

kwasowsky, 2024-12-06:
Bugs descriptions:
1. iCUE LINK XD5 Elite LCD - On Windows in iCue i set my screen to show me GIF. It still shows me this GIF (it was set to show GIF for device setting, not only if iCue software is running). Doesn't matter what I choose - seems like setting doesn't apply at all.
2. XC7 ELITE LCD - Same thing as above with a GIF device setting but this one after installation of OpenLinkHub changed to Liquid temp - it didn't change my setting of 180° rotation though. I can't change nothing except LEDs of the ring around LCD.

Expected behavior:
Settings to apply properly.

My devices + firmware versions:
1. 2x iCUE LINK System Hub (firmware v.2.9.488)
2. iCUE LINK XD5 Elite LCD (it doesn't say which firmware is used)
3. XC7 ELITE LCD (firmware v.0.0.0.19)
4. 6x iCUE LINK QX RGB (fans don't have firmware if I'm correct)
5. Dominator Titanium RGB 32GB DDR5 (2x16GB) 7200Mhz (CMP32GX5M2X7200C34) (it doesn't show up in dashboard - I've read that this is by default)
6. MM700 RGB mousepad (firmware v.1.0.5)
7. Nightsabre wireless mouse + USB dongle (it doesn't show up in dashboard - I think it's not supported)

I checked on:
 - OS: Linux Arch / EndeavourOS;
 - Browser: Brave (v.1.73.93) / Firefox (v.133);

Suggestions:
- choosing colors of static led colors;
- changing FPS on LCD screens;
- enabling RAM by default for newbies (please explain is this security or other reason?);
- 2500rpm for pump is default on quiet profile. 2500rpm it's not quiet... (I'm aware of custom profile setting);
- custom profile settings - min. pump speed is 50% whis is not quite and too high for idle. imo should be 20-30 to be completely silent and still working fine. If I set lower than 50% after refreshing page it gets back to 50%;
- 33%/66%/100% only for brightness settings - slider with 1% or 5% change between would be great;
- Changing speed of fans depending on GPU/CPU temps, not only CPU or GPU. If one or another get higher temps then faster rpm speeds are applied (that is something that even iCUE doesn't have and I wanted it really bad);

Thank you for providing us software for Corsair iCUE hardware on Linux and for your hard work!
If I missed something that is already implemented or cannot be for whatever reason please let me know.

jurkovic-nikola, 2024-12-06:
Hello, thank you for detailed description, really appreciate it. 

Bugs:
1. What you set in ICUE will only run when iCUE is running. What you set in Device Mode, will only run until other software takes control. So, your image will be replaced with default one from OpenLinkHub while software is running. Once software is turned off, LCD goes back to default Device Mode. GIF can be implemented tho. 
2. Correct, and its fixed. Will be released with 0.3.9. You'll be able to chnage LCD mode and rotate

Suggestions:
- RGB editor is currently in planning phase for each device. For time being, static color can be changed in /opt/OpenLinkHub/database/rgb/your-device-serial.json. Serial is end path of link in the Web Dashboard. 
- Until any animations are implemented, there is no need for this. Currently LCD is updated once per second. 
- Not possible due to SMBUS not being visible on some boards. This needs to be done manually by user and user needs to understand what `acpi_enforce_resources=lax` is. 
- This is also fixed and will be released in 0.3.9. This setting and limitation was carried over from an AIO implementation and was never changed for XD5 devices. 
- Same as above. 
- Brightness setting was implemented same as one in iCUE, you have 2 stages, 33, 66 and 100%. Even tho you have a slider, it only drops by 1/3. Gradual +-5% can be implemented. 
- Yeah, that's possible since program already have all the data, its just logic implementation.

kwasowsky, 2024-12-08:
Bugs annot.:
1. Yes that's correct, only problem is that in iCUE software you can't change or turn off colored LED ring around CPU block LCD. That's a bummer. Every thing else I could set as I want it in software or device mod. Otherwise I wouldn't need software for Linux if it comes to looks (and most important part - I don't think cooling and profile works in device mod - most important thing and Corsair didn't implement it...).
2. That's great. Can't wait to try it out!

Suggestions annot.:
1. I am new to Linux but all I found in /OpenLinkHub/database/rgb is .gitkeep file. In database folder there is rgb.json
2. True. Ok.
3. I understand. When I have more time I will look into it.
4. Good to know, thank you.
5. ~||~
6. That was my bad and misunderstanding. I was talking about brightness of LEDs, not LCDs. In iCUE LCDs brightness can be set to 33/66/100. I was thinking about RGB brightness but I guess this can be set in RGB settings - for ex. instead of 255/255/255 - 50/50/50 which is same color but dimmed.
7. That would be awesome to see it one day. I don't know why Corsair didn't think about that in the first place.

jurkovic-nikola, 2024-12-08:
Lets talk over Discord. https://discord.gg/2NUN43U4

kwasowsky, 2024-12-08:
1. Is this OS specific folder? I don't see 'opt' directory in my filesystem. I found OpenLinkHub directory in home/OpenLinkHub/database/rgb and this folder only contains hidden file '.gitkeep'.
2. Nice. I just checked it out. Workes quiet and nice with rpm.
3. Great! Now I can use it without getting blind at night haha! A text input for 1-100 value next to it would be great for OCD people.
4. That CPU+GPU is huge! That is the best feature for people running whole system in one loop like me.

Good to know.

Yes. I see that Dashboard uses 'fonts.gstatic.com' and 'fonts.googleapis.com'. Although would be great if dashboard didn't connect to anything online. For ex. for people who has blocked google addresses blocked or offline users.

FYI:
Just so you know. My XD7 LCD has 180 degree in device setting because I had to mount it like that. With this device setting I have to keep standard setting in OpenLinkHub. If I set up 180 degree in OpenLinkHub it sets up like default. So I guess your software reads device settings and to that adds +180 degree.

Waiting for XD5 pump LCD screen implementation. Still can't setup screen.

Thank you for fixes!

P.S.
I'm running temp OS right now I have a temporary break from Discord but I joined. Thanks for the link!

jurkovic-nikola, 2024-12-08:
Hey, what exactly is the issue with XD5 ? Not being able to setup GIF or LCD does not work at all ? Just to know in which direction to go.

kwasowsky, 2024-12-08:
My pump is still showing my GIF from device setting. Whatever I choose in OpenLinkHub it doesn't apply.

jurkovic-nikola, 2024-12-08:
I'll need output of `lsusb` command to see what's the device ID.

jurkovic-nikola, 2024-12-08:
Grab 0.4.0. This will also fix rotation issue.

kwasowsky, 2024-12-08:
I just checked it out. Both LCD screens work as intended. I managed to change json values to match my colors of preference. Thank you very much.