jurkovic-nikola/OpenLinkHub #35: I need help

Angelias1231, 2024-12-08:
HI!!! I got a Question??? I am new to Linux and i am using Pop_OS .

Went i try to install at the time to enter the command ( go build . ) it tell me ( go.mod:3: invalid go version '1.22.2': must match format 1.23 ) 

I try to find i way to update go but i don't find anything that tell me how to do it. 

can you please help me???   thank you for your time

jurkovic-nikola, 2024-12-08:
Hello, you need to have at least 1.22.2 go version. I've just tested on fresh installation of PopOS. Make sure you don't have any previous version of go installed via apt or other package manager. 

1. Download GO
```bash
wget https://go.dev/dl/go1.22.2.linux-amd64.tar.gz
```
3. Extract GO
```bash
tar -xvf go1.22.2.linux-amd64.tar.gz
```
4. Move extracted GO to `/usr/local`
```bash
sudo mv go /usr/local/
```
5. Edit your shell environment file (bash example)
```bash
vim ~/.bashrc
```
6. Add at the end of the file. 
```bash
export GOROOT=/usr/local/go
export GOPATH=$HOME/go
export PATH=$GOPATH/bin:$GOROOT/bin:$PATH
```
7. Update shell
```bash
source ~/.bashrc
```

closed 2024-12-08 by jurkovic-nikola

Angelias1231, 2024-12-08:
Hi!!!! thank for the help with the Go install. I did what was said and it give me this at the end in the terminal 

angel@pop-os:~$ cd OpenLinkHub/
angel@pop-os:~/OpenLinkHub$ go build .
angel@pop-os:~/OpenLinkHub$ chmod +x install.sh
angel@pop-os:~/OpenLinkHub$ sudo sh install.sh
[sudo] password for angel: 
Checking if application username openlinkhub exists...
Application username openlinkhub found. Skipping username creation...
Installation is running...
id: ‘openlinkhub’: no such user
Setting permissions...
chown: invalid user: ‘openlinkhub:openlinkhub’
angel@pop-os:~/OpenLinkHub$ 

do you have a ideal what i did wrong  Thank you for your time for helping me out and your time you spend on this project