jurkovic-nikola/OpenLinkHub #41: triple 6 perms on hidraw device

jamcinnes2, 2024-12-13:
**Is your feature request related to a problem? Please describe.**
Is it necessary to give rw (666) permissions to the hidraw keyboard device? There are udev rules specifying mode 0666 for my keyboard. I'm ignorant maybe this is just fine... ?

**Describe the solution you'd like**
Concerned about giving rw to everyone.

**Describe alternatives you've considered**
Restrict mode on hidraw device.

**Additional context**
Add any other context or screenshots about the feature request here.

jamcinnes2, 2024-12-13:
p.s. this is awesome, my LEDs work great

jurkovic-nikola, 2024-12-14:
Hello,

Thank you for reporting. Permissions are changed to 600 and locked to `openlinkhub` username.

closed 2024-12-14 by jurkovic-nikola