jurkovic-nikola/OpenLinkHub #42: Commander Pro Hub not shown anymore

pyrotiger, 2024-12-18:
Both of my Commander Core XT and PRO hubs used to work flawlessly out of the box at install, after a recent update. That functionality broke, i was able to run device permissions script, that gave me the Core XT back, but the Pro hub is still not showing up in the dashboard as expected. 

https://forum.garudalinux.org/t/openlinkhub-not-detecting-corsair-devices/40721
Garuda Linux Zen Kernel 6.12.4

jurkovic-nikola, 2024-12-18:
Hello,

Include stdout.log in the report. 

Maybe I'm missing something but there is no Commander Pro in your lsusb output.

pyrotiger, 2024-12-19:
im sorry Commander Core for my h150 Elite Capellix and Commander Core XT. i fixed that by running the /device-permissions.sh script found in the folder. However, both devices now show but i cannot change the picture of the LCD anymore like i used to be able to. I'm also now seeing this error in my jornal error instead of the USB2 error output.
~~~
Dec 18 06:50:52 PHOENIXATE systemd-udevd[444]: /etc/udev/rules.d/99-corsair-0c2a.rules:1 Unknown user 'openlinkhub', ignoring.
Dec 18 06:50:52 PHOENIXATE systemd-udevd[444]: /etc/udev/rules.d/99-corsair-0c32.rules:1 Unknown user 'openlinkhub', ignoring.
Dec 18 06:50:52 PHOENIXATE systemd-udevd[444]: /etc/udev/rules.d/99-corsair-0c39.rules:1 Unknown user 'openlinkhub', ignoring.
Dec 18 06:50:52 PHOENIXATE systemd-udevd[444]: /etc/udev/rules.d/99-corsair-1b7c.rules:1 Unknown user 'openlinkhub', ignoring.
Dec 18 06:50:52 PHOENIXATE systemd-udevd[444]: /etc/udev/rules.d/99-corsair-1b7f.rules:1 Unknown user 'openlinkhub', ignoring.
~~~

pyrotiger, 2024-12-19:
here is my stdout.log, the file itself is 101mb so i can't upload here, there is more to the output but it is repetitive of the last few lines. In the past, I tried to set it to manipulate my RAM as instructed but it does not work and gave up on that one.

~~~
{"file":"/home/pyro/Downloads/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is missing, creating initial one.","time":"2024-12-16T17:15:24-07:00"}
{"file":"/home/pyro/Downloads/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is created.","time":"2024-12-16T17:15:24-07:00"}
{"file":"/home/pyro/Downloads/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is missing, creating initial one.","time":"2024-12-16T17:15:24-07:00"}
{"file":"/home/pyro/Downloads/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is created.","time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7039,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3114,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3114,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3129,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3122,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3122,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-16T17:15:24-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-16T17:15:24-07:00"}
{"error":"listen tcp 127.0.0.1:27003: bind: address already in use","level":"fatal","msg":"Unable to start REST server","time":"2024-12-16T17:15:24-07:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-18T13:49:59-07:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-18T13:49:59-07:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2024-12-18T13:49:59-07:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3122,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3122,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-18T13:49:59-07:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7036,"time":"2024-12-18T13:49:59-07:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/b10b20708a4884bab08fc3266091005f-ee.json","msg":"Loaded custom user profile","serial":"b10b20708a4884bab08fc3266091005f","time":"2024-12-18T13:49:59-07:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/b10b20708a4884bab08fc3266091005f.json","msg":"Loaded custom user profile","serial":"b10b20708a4884bab08fc3266091005f","time":"2024-12-18T13:49:59-07:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/b10b20708a4884bab08fc3266091005f-ee.json","msg":"Loaded custom user profile","serial":"b10b20708a4884bab08fc3266091005f","time":"2024-12-18T13:50:00-07:00"}
[... cut ...]
{"error":"No such device","level":"error","msg":"Unable to write to endpoint","serial":"205F30764B4B","time":"2024-12-18T17:50:28-07:00"}
{"error":"No such device","level":"error","msg":"Unable to write to a device","serial":"205F30764B4B","time":"2024-12-18T17:50:28-07:00"}
{"error":"No such device","level":"error","msg":"Unable to write to endpoint","serial":"205F30764B4B","time":"2024-12-18T17:50:28-07:00"}
{"error":"No such device","level":"error","msg":"Unable to write to a device","serial":"205F30764B4B","time":"2024-12-18T17:50:28-07:00"}
~~~

jurkovic-nikola, 2024-12-19:
@pyrotiger Thanks for the log. Application needs to run under openlinkhub username. 

Upgrade to latest 0.4.3 version if you did not already. 
1. Create username
```bash
sudo useradd -r "openlinkhub" --shell=/bin/false
```
2. Folder permissions
```bash
chown -R "openlinkhub":"openlinkhub" /opt/OpenLinkHub/
```
```bash
chmod -R 755 /opt/OpenLinkHub/
```
3. Restart application
```bash
sudo systemctl start OpenLinkHub
```
Clear your browser cache due to changes in JS files for LCD. Regarding memory, join to Discord https://discord.gg/Bhz5pj3K for more info.

pyrotiger, 2024-12-19:
I did as you suggested, still getting the same journal error.

While waiting for a reply, I went ahead and removed the installation and reinstalled openlinkhub. The LCD issue has disappeared, but the unknown user error persists.

jurkovic-nikola, 2024-12-21:
Issue is resolved over Discord.

closed 2024-12-21 by jurkovic-nikola