jurkovic-nikola/OpenLinkHub #43: RGB power defaults to 50% at power on

iocc-git, 2024-12-26:
**Describe the bug**

The state of amount of light (was 33%,66%,99% before) it emits isnt saved
when the RGB are turned off and later on. It always defaults to 50% power.

**Expected behavior**

Save the current value and restore it.

**Desktop (please complete the following information):**

AlmaLinux 9
1.6.336
0.4.3

jurkovic-nikola, 2024-12-26:
Hello, correct. Current scheduler turns RGB off and later on restore RGB to default 50% value. New release will store current brightness value and will restore to that value.

iocc-git, 2024-12-27:
Thanks.
Brightness was the word that I didnt find yeasterday.
Can I also suggest a brightness icon instead of that flash? It confuses me.
Example of an brightness icon big and small that everyone can understand: https://i.sstatic.net/brcaB.png
[image]

jurkovic-nikola, 2025-01-02:
This is implemented in 0.4.4.

closed 2025-01-02 by jurkovic-nikola