jurkovic-nikola/OpenLinkHub #44: Display running version

iocc-git, 2024-12-27:
**Describe the solution you'd like**

http://127.0.0.1:27003/ says alot, like mine:

`
OS: Linux               Build: ab158377cadde2c0295757f314b0d7e460442c08         Intel(R) Core(TM) i9-14900KF (24C - 32T)
Architecture: x86_64    2024-12-17 11:22:58 +0000 UTC                           System Product Name (1641 - 02/21/2024)
`

Somewhere here, can the version number of OpenLinkHub be added?

**Additional context**

Build isnt very usefull for me, the version would be much more relevant.

jurkovic-nikola, 2024-12-27:
`Build: ab158377cadde2c0295757f314b0d7e460442c08` directly links to git commit. There is no point of making custom versioning when GO already makes this automatically on binary build.

closed 2024-12-27 by jurkovic-nikola

iocc-git, 2024-12-28:
my build doesnt link to anything.
and only way to figure out what version that is installed is to do rpm -q OpenLinkHub

jurkovic-nikola, 2024-12-28:
X.X.X version is only used for making packages. Version needs to be incremented when you're upgrading. 

Actual build version is under Build section. [ab158377cadde2c0295757f314b0d7e460442c08](https://github.com/jurkovic-nikola/OpenLinkHub/commit/ab158377cadde2c0295757f314b0d7e460442c08). This program won't open any external link except GitHub logo at top right corner.

iocc-git, 2024-12-30:
Its not the build version that is asked for when you make a bugreport, its the X.X.X version.
I dont understand why it would be a negative thing to make it harder to get when reporting bugs and so on.

jurkovic-nikola, 2025-01-02:
This is implemented in 0.4.4.