jurkovic-nikola/OpenLinkHub #45: (Help) Cannot get running on Linux Mint Cinnamon 22

JustaFloatingEyeYT, 2024-12-30:
**Describe the bug**
When I install the .deb `sudo apt deb [.deb]`, it installs (and also makes my fans go from white to nothing) but cannot find it in the start menu. After trying to find the apps location (/opt/OpenLinkHub), trying to run it through terminal `./OpenLinkHub` brings up a bunch of text that I do not understand, seems like errors.

`{"level":"info","msg":"Failed to log to file, using default stderr","time":"2024-12-29T01:56:43Z"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-29T01:56:43Z"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-29T01:56:43Z"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2024-12-29T01:56:43Z"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-29T01:56:43Z"}
{"error":"Failed to open a device with path '/dev/hidraw0': Permission denied","level":"error","msg":"Unable to open HID device","productId":3114,"time":"2024-12-29T01:56:44Z","vendorId":6940}
[Server] Running REST and WebUI on 127.0.0.1:27003. WebUI is accessible via: http://127.0.0.1:27003
{"error":"listen tcp 127.0.0.1:27003: bind: address already in use","level":"fatal","msg":"Unable to start REST server","time":"2024-12-29T01:56:44Z"}`

**To Reproduce**
Steps to reproduce the behavior:
1. Download the .deb release
2. Install it using apt

**Expected behavior**
1. Download the .deb release
2. Install it using apt
3. Have it appear in the start menu, and launch fine

**Desktop (please complete the following information):**
 - OS: Linux Mint Cinnamon 22
 - Device Firmware: Custom built PC, AMD CPU/GPU, Samsung SSD, MSI Motherboard, Corsair case/fans/fanhub
 - Program Version Latest

jurkovic-nikola, 2024-12-30:
Hello,

Running it manually with your user will not work due to device permissions. Please upload stdout file in issue.

JustaFloatingEyeYT, 2024-12-30:
Ah! So I was being silly then and didn't read the instructions correctly. Well now that I found out it is used in web, I can say it works.

closed 2024-12-30 by JustaFloatingEyeYT