jurkovic-nikola/OpenLinkHub #47: (RGB/Lighting) AF Series fan support?

JustaFloatingEyeYT, 2024-12-30:
**Describe the solution you'd like**
So I noticed there's an option for lighting in the app. Great! I can finally use my fan lights (they were white by default and when I installed the app they turned off) as well as change the fan speeds of course. But I notice that none of the fan options work as my fans are an AF series and that is not one of the options. 

So will there be support for these fans or am I doing something wrong?

jurkovic-nikola, 2024-12-30:
What controller do you have to control them ? Send me lsusb output

JustaFloatingEyeYT, 2024-12-30:
[image]
I'm not sure if this is what you mean by controller but its this fanhub here.

And I am not sure where I find the lsusb output?

jurkovic-nikola, 2024-12-30:
Is your fan rgb cable 3 pin or 4 pin ?

JustaFloatingEyeYT, 2024-12-30:
4 pin

jurkovic-nikola, 2024-12-30:
Are they directly connected to Commander Core XT RGB PPRTA or you're using Lighting Node connected to XT ?

JustaFloatingEyeYT, 2024-12-30:
i think im using the built in rgb hub with the 5000d (case im using)

jurkovic-nikola, 2024-12-30:
Send me image of how it's connected in the case. 4pin rgb fans are automatically detected if connected directly to Commander Core XT

JustaFloatingEyeYT, 2024-12-30:
[image]

JustaFloatingEyeYT, 2024-12-30:
The 3 fans at the top do connect to the fan hub too I am certain. as the app detects 4 fans.

jurkovic-nikola, 2024-12-30:
Yeah, thats Corsair RGB Hub. Set on Commander Core XT page 3 pin Corsair hub fans to 8 led fan and set amount. This should activate the rgb on fans to default blue color.

JustaFloatingEyeYT, 2024-12-30:
I am not sure if the cables could reach but ill try that

JustaFloatingEyeYT, 2024-12-30:
[image]
They are working and light up!

How do I change the colour though?

jurkovic-nikola, 2024-12-31:
There is file for each device in dstabase/rgb folder jn /opt/OpenLinkHub. Colors are changed there until RGB editor is implemented.

JustaFloatingEyeYT, 2024-12-31:
Alright, thanks!

closed 2024-12-31 by JustaFloatingEyeYT