jurkovic-nikola/OpenLinkHub #48: Cannot detect LEDs

mosquitodawg, 2024-12-31:
Thanks for this project!

I installed the project automatically using apt install.

As you can see it is not detecting 3 fans LED from my 4000d corsair case. The fans are AF120 RGB ELITE Fans.

[image]

Here is the connections of the hubs and motherboard.

[image]

thanks!

jurkovic-nikola, 2024-12-31:
Can you provide `/opt/OpenLinkHub/stdout.log` content ? This appears like a permission issue, since Lighting Node Pro is not even picked up by the application.

mosquitodawg, 2024-12-31:
Yeah, here is the contents:

```
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is missing, creating initial one.","time":"2024-12-30T14:38:07-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is created.","time":"2024-12-30T14:38:07-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is missing, creating initial one.","time":"2024-12-30T14:38:07-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is created.","time":"2024-12-30T14:38:07-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3083,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:08-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T14:38:14-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:38:14-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2024-12-30T14:38:14-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:38:14-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3083,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T14:38:15-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T14:43:29-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:43:29-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T14:45:37-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:45:37-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T14:53:16-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:53:16-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T14:53:31-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T14:53:31-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2024-12-30T17:50:48-05:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T17:50:48-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2024-12-30T17:50:48-05:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":3083,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T17:50:48-05:00"}
[... cut ...]
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T19:13:34-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T19:13:34-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T19:13:34-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T19:13:34-05:00"}
{"error":null,"level":"warning","msg":"Invalid permissions","productId":7165,"time":"2024-12-30T19:13:34-05:00"}
```

jurkovic-nikola, 2025-01-02:
Hello, upgrade to latest 0.4.4 version. Also, join to Discord if you need any help regarding device permissions. https://discord.gg/SuJ7h5ym

closed 2025-01-02 by jurkovic-nikola