jurkovic-nikola/OpenLinkHub #50: error 500 "unable to serve web content"

cnschu, 2025-01-02:
**Describe the bug**
Only 2 of 7 fans are displayed.

**To Reproduce**
1. navigating to http://127.0.0.1:27003
2. click on "iCUE LINK System HUB"
3. only 2 of my 7 fans are displayed, the third displays an error 500 (see screenshot)
[stdout.log](https://github.com/user-attachments/files/18292347/stdout.log)

**Expected behavior**
all fans should be displayed

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Linux Mint 22
 - Device Firmware: 2.11.517
 - Program Version: unknown, as there is no version displayed in the GUI

jurkovic-nikola, 2025-01-02:
Hello, remove `DevicePosition` field in your profile and upgrade to 0.4.4 version. This is caused by wrong deviceId enumeration and channelId does not exists. 

Edit `/opt/OpenLinkHub/database/profiles/serial-number.json` file and remove `DevicePosition` field from JSON and restart service. Program will automatically re-create this with device serial numbers that are static.

cnschu, 2025-01-02:
there is no DevicePosition in the JSON, did you mean "Positions"?

jurkovic-nikola, 2025-01-02:
You.

cnschu, 2025-01-02:
works

closed 2025-01-02 by cnschu