jurkovic-nikola/OpenLinkHub #52: SImple Install for BazziteOS

chunky-shrapnel, 2025-01-09:
Is there any easy or simple way to build this out for a BazziteOS (Fedora) based system. 

I feel like with the increasing amount of custom PC's starting to run SteamOS based systems, this software might need a simple install for corsair based Linux builds. 

Is this possible at all?

ArandomHitman, 2025-01-10:
I am not familiar with BazziteOS, but is there any way to run it as one of the file types in the release channel?

jurkovic-nikola, 2025-01-10:
Remove any previous version
```shell
sudo rpm-ostree uninstall OpenLinkHub
```
Download 0.4.7 RPM
```shell
wget https://github.com/jurkovic-nikola/OpenLinkHub/releases/download/0.4.7/OpenLinkHub-0.4.7-1.x86_64.rpm
``` 
Install 0.4.7 version
```bash
sudo rpm-ostree install /path/to/filename/OpenLinkHub-0.4.7-1.x86_64.rpm
```
Reboot
```shell
systemctl reboot
```

closed 2025-01-10 by jurkovic-nikola

chunky-shrapnel, 2025-01-13:
Okay great installed, now how does one open it from there?

jurkovic-nikola, 2025-01-13:
Open what ? Application runs as a service and configuration is done thru web interface in your browser. `http://127.0.0.1:27003/`

chunky-shrapnel, 2025-01-13:
I think that's where I'm getting confused, I though it was more like a regular app that opens like a normal flatpak (would his be possible in the future?). Regardless, the ip sends an unable to connect error in firefox.

jurkovic-nikola, 2025-01-13:
No, its web based configuration. Please join to Discord https://discord.gg/exWf3zMP