jurkovic-nikola/OpenLinkHub #53: Support for Virtuoso XT RGB Wireless

kronflux, 2025-01-09:
The lack of linux support for something as simple as being able to turn off the Mic LED on my headset is the literal only reason I've held off on switching to Linux for well over 2 years now. I am getting really tired of Windows, and I desperately want to switch, but I don't want to be blinded by my mic.

Is there any way to get this device supported?
It seems that ckb-next is very much at a stand-still when it comes to new device support.

I would be more than happy to test any configuration, on any OS, with whatever data you need collected.
For reference, I have used ckb-next to output reports for both wireless and wired:
[ckb-next-report-1736411264-wireless.gz](https://github.com/user-attachments/files/18358653/ckb-next-report-1736411264-wireless.gz)
[ckb-next-report-1736411352-wired.gz](https://github.com/user-attachments/files/18358654/ckb-next-report-1736411352-wired.gz)
hopefully the information in these reports is enough to get the ball rolling if you're willing.

jurkovic-nikola, 2025-01-10:
Update to 0.4.7

closed 2025-01-10 by jurkovic-nikola

kronflux, 2025-01-10:
Hey!
First of all, let me say THANK YOU!!!!! This is.... life changing!

Secondly, I just tested this out, it does sort of work, but the brightness slider controls both the RGB on the sides, as well as the mic LED at the same time, rather than being separate.
When you hit the mute button, it changes to red, as one would expect, but when you press it to unmute, it changes to Green, regardless of what colors were chosen.
Changing the brightness slider again reverts it back to the original color.

Basically what I'm looking for is the ability to turn off the LED when the mic is unmuted, and be red(or custom color) when muted.
The brightness of the mic LED isn't important, but it should be separate from the LEDs on the sides.

Anything I can provide to get this device more polished? :)

jurkovic-nikola, 2025-01-10:
Set your microphone to Black (0,0,0). 

When you press mute, it will go Red for a 1.5 second, then goes back to original value in profile you set. When you press again mute, it will unmute and green color will be displayed again for 1.5 second, and will revert back to original value in the profile. 

[image]

jurkovic-nikola, 2025-01-10:
Sure, i'll add it for 0.4.8.

kronflux, 2025-01-10:
Thank you so much ❤️
I can finally switch to Linux as my primary OS! 😁

Quick note, indicator LED doesn't seem to save, it reverts to the default light blue when you change it, I'm not sure if this is intentional or not, but figured I'd mention :)

jurkovic-nikola, 2025-01-10:
Heh, it happens only on Wireless mode. You can fix it if it buggs you by editing html file `/opt/OpenLinkHub/web/virtuosorgbXTW.html`. Change:
`<input type="hidden" id="zones" name="zones" value="2">` to `<input type="hidden" id="zones" name="zones" value="3">` and restart service.

kronflux, 2025-01-10:
Awesome! Tried it, and it's working! Thanks again! Any way the community can support you? sponsor or donate, or anything like that? This is actually game changing!

jurkovic-nikola, 2025-01-10:
No, this project does not accept any donations.