jurkovic-nikola/OpenLinkHub #54: [Feature Request] - better animated screens for icue LINK LCD devices

ArandomHitman, 2025-01-09:
**Describe the solution you'd like**

Having recently transitioned from Linux to Windows, I've had nothing but issues with other programs in managing my Corsair ICUE LINK devices. Thanks to your program, I was able to fix 99% of all issues I had on Linux, but I do have one question: Is it possible to actually use LCD device features (e.g., the XD5 LCD, which I have myself) and properly animate the screens? 

If needed, I can try helping with this. I've had some experience with Javascript and GO, alongside a decent bit of experience with HTML/CSS. I don't know exactly what language you've used for the LCDs here, but if you could let me know your thoughts, that would be great.

jurkovic-nikola, 2025-01-09:
LCD is already animated, you just need to put your gif image `/opt/OpenLinkHub/database/lcd/images/`. Images are loaded on program startup so you'll have to restart service. 

After that change your LCD mode to Image / GIF
[image]

ArandomHitman, 2025-01-10:
I should've cleared this up. My apologies.

I guess this is more of a nitpick, but with ICUE you can customize background animations for temperature statistics that you put on LCD screens. This is what I meant, apologies.

jurkovic-nikola, 2025-01-10:
This was tried before and has major impact on your CPU, so idea was scrubbed. There are couple of other ways to pre-cache all animation effects, but currently this is not in focus as basic functionality exists. Maybe down the project line this will get implemented once most of devices have basic functionality. It's also unclear what type of animations to implement (no iCUE stuff).

closed 2025-06-15 by jurkovic-nikola