jurkovic-nikola/OpenLinkHub #55: Changing color of static rgb

james-math, 2025-01-10:
Is there any way to change the color of the static preset?  Right now it's set to cyan and I can't find any way through the dashboard or API to change it to another color.  I'm using an icue link system hub with rx120 rgb fans and a titan 240 aio.  I'd be interested more generally in being able to modify the parameters of the other presets too if that's a possibility.

gitToast27, 2025-01-10:
Saw this hope this helps as its what I do. 
/opt/OpenLinkHub/database/rgb 
Search for static set values of rgb, then use the command systemctl restart OpenLinkHub.service (restarts service)
Should be able to do this rgb editor as well on the new version. 

PS. If your familiar the app midnight commander makes this a bit more fluid.

james-math, 2025-01-10:
Thanks, that works!  I'll experiment with this to see if there's any way to control the other animations.  I still hope that there can be some way to incorporate this into the dashboard, but I'm glad it's doable.

jurkovic-nikola, 2025-01-10:
You have RGB editor in latest version in Dashboard.

closed 2025-01-10 by jurkovic-nikola