jurkovic-nikola/OpenLinkHub #56: [Feature Request] - Support for the upcoming Corsair XENEON EDGE

ArandomHitman, 2025-01-10:
**Describe the solution you'd like**

**Additional context**
The Xeneon Edge is an upcoming Corsair product that is not strictly tethered to ICUE but serves as a separate monitor that will work with Windows and MacOS. It is set to release sometime in the next month or so. Hence, I am asking for you to consider adding support for it when it does come out in the future, for now you can ignore this as it's not a priority.

More info can be found here:
https://www.corsair.com/uk/en/explorer/gamer/monitors/corsair-xeneon-edge-touchscreen-everything-you-need-to-know/ 
https://www.techpowerup.com/330902/corsair-xeneon-edge-touchscreen-is-your-monitors-copilot/
https://www.youtube.com/shorts/PUe827vHQgs

jurkovic-nikola, 2025-01-10:
Hello, such devices have no actual software control since they are marked as monitors. Everything you do on them are managed thru screen or another software like AIDA etc. I'll leave this open until some review is made public.

Tavus1990, 2026-04-02:
Is the Xeneon Edge now supported or not ?,im on the fence to buy one.

jurkovic-nikola, 2026-04-02:
Its not supported. Probably in some next version kiosk web overview will be made, but thats it.

AlasdairWilkie, 2026-09-12:
Would love some support for MacOS

lunetics, 2026-10-22:
Hey, i build a tool to control the xeneon via DDC/USB (Including OSD) , maybe it helps for OpenLinkHub: https://github.com/lunetics/corsair-xeneon-edge-ctl

closed 2026-11-01 by jurkovic-nikola