jurkovic-nikola/OpenLinkHub #60: Device not located

CarterCakes, 2025-01-18:
Hello,
My H100 water pump isn't being detected by this, so I am not able to locate the RGB settings. In fact it seems like I do not have the Icue link system hub on here. Is there a resolution for this?

[image]

[image]

jurkovic-nikola, 2025-01-18:
Is your device in the supported list ? Please paste `lsusb` output from terminal.

CarterCakes, 2025-01-18:
Yes its the iH100 RGB elite. Disregard the link system hub, I am just hoping to focus on the water cooler RBG for now.

jurkovic-nikola, 2025-01-18:
Please join to Discord https://discord.gg/exWf3zMP

H devices without I at the end usually don't have any USB control, and they are controlled via PWM from MOBO.

CarterCakes, 2025-01-18:
It does have the I at then end, but I am joining the discord now, thank you!

jurkovic-nikola, 2025-01-18:
Resolved over Discord.

closed 2025-01-18 by jurkovic-nikola