jurkovic-nikola/OpenLinkHub #61: Manjaro with Arch Package - USB device not detected

EnerGeh, 2025-01-21:
Hi, I looked and googled a lot in the last couple of weeks but I didn't find any solution for that: I use Manjaro and installed the according Arch Package from AUR ([here](https://aur.archlinux.org/packages/openlinkhub)). I use a K100 RGB (1b7c).
Unfortunately, there is no device detected when starting the service:
``● openlinkhub.service - Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs
     Loaded: loaded (/usr/lib/systemd/system/openlinkhub.service; enabled; preset: disabled)
     Active: active (running) since Tue 2025-01-21 08:08:18 CET; 9h ago
 Invocation: 9051edd78df14e43a702b474c06a550d
   Main PID: 1123 (openlinkhub)
      Tasks: 17 (limit: 38358)
     Memory: 60.7M (peak: 72.5M)
        CPU: 2.666s
     CGroup: /system.slice/openlinkhub.service
             └─1123 /usr/bin/openlinkhub

jurkovic-nikola, 2025-01-21:
Check content of stdout.log file

jurkovic-nikola, 2025-01-21:
After further troubleshooting on Discord, keyboard is never registered as hidraw device on the system, so there is nothing for program to communicate with. After that issue is solved, program will work just fine.

closed 2025-01-21 by jurkovic-nikola