jurkovic-nikola/OpenLinkHub #62: Separate static from dynamic state files

Tyler-Hardin, 2025-01-23:
**Describe the solution you'd like**
It would be nice if the web/ and static/ could be stored in a separate dir from database/ et al. I'd like to be able to pass the root directories for static and dynamic dirs as an option.

**Additional context**
I'm trying to package this app for NixOS. I have succeeded, but the solution is quite dirty. In Nix, it's idiomatic to put static files associated with a given package into (what is effectively equivalent to) /usr/share/OpenLinkHub and to put dynamic state into /var/lib/OpenLinkHub. Since OLH expects the static files to exist in the working dir, I've had to do this with an overlayfs.

This is what I'm doing now.
```
{ config, lib, pkgs, ... }: {
  options = {
    openlinkhub.enable = lib.mkEnableOption "OpenLinkHub";
  };

  config = lib.mkIf config.openlinkhub.enable {
    nixpkgs.overlays = [
      (final: prev: {
        openlinkhub = prev.buildGoModule {
          name = "OpenLinkHub";
          src = prev.fetchFromGitHub {
            owner = "jurkovic-nikola";
            repo = "OpenLinkHub";
            rev = "3f21efe11503456f522b9494c75be4ff1064123c";
            sha256 = "sha256-X/ck5OmRgp4BY6NokA7Ua7Q0uTagq5PEXtpUxoIERJA=";
          };
          vendorHash = "sha256-o+Ek5pJG4fjeQg8e11afAbVIvzmDvG92v2mp4yOM3x4=";

          buildInputs = with prev; [ systemdLibs pciutils ];
          postInstall = ''
            mkdir -p $out/share/OpenLinkHub
            cp -r database $out/share/OpenLinkHub
            cp -r static $out/share/OpenLinkHub
            cp -r web $out/share/OpenLinkHub

            mkdir -p $out/lib/udev/rules.d/
            cp *.rules $out/lib/udev/rules.d/
            substituteInPlace $out/lib/udev/rules.d/*.rules --replace-warn "/bin/sh" "${prev.bash}/bin/sh"
          '';
          meta.mainProgram = "OpenLinkHub";
        };
      })
    ];

    users.users.openlinkhub = { isSystemUser = true; group = "openlinkhub"; };
    users.groups.openlinkhub = {};

    # unnecessary
    systemd.mounts = [{
      what = "overlay";
      type = "overlay";
      where = "/var/lib/OpenLinkHub/merged";
      options = "lowerdir=${pkgs.openlinkhub}/share/OpenLinkHub/,upperdir=/var/lib/OpenLinkHub/upper,workdir=/var/lib/OpenLinkHub/work";
    }];

    systemd.services.OpenLinkHub = {
      description = "Open source interface for Corsair iCUE LINK hubs, fans, and AIOs";
      after = [ "sleep.target" ];
      wantedBy = [ "multi-user.target" ];
      script = "${lib.getExe pkgs.openlinkhub}";
      reload = "kill -s HUB $MAINPID";
      path = [ pkgs.pciutils ];

      serviceConfig = {
        User = "openlinkhub";
        Group = "openlinkhub";
        RestartSec = 5;
        StateDirectory = "OpenLinkHub";
        WorkingDirectory = "/var/lib/OpenLinkHub/merged";
        ReadWritePaths = [
          "/var/lib/OpenLinkHub/upper"
          "/var/lib/OpenLinkHub/merged"
          "/var/lib/OpenLinkHub/merged/database/"
          "/var/lib/OpenLinkHub/merged/database/profiles"
          "/var/lib/OpenLinkHub/merged/database/rgb"
          "/var/lib/OpenLinkHub/merged/static/"
          "/var/lib/OpenLinkHub/merged/web/"
        ];
      };
    };

    # unnecessary
    systemd.tmpfiles.rules = [
      "d /var/lib/OpenLinkHub 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/upper 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/upper/database 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/upper/database/profiles 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/upper/database/rgb 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/work 0755 openlinkhub openlinkhub - -"
      "d /var/lib/OpenLinkHub/merged 0755 openlinkhub openlinkhub - -"
    ];

    environment.systemPackages = [ pkgs.openlinkhub ];

    services.udev.packages = [ pkgs.openlinkhub ];
  };
}
```

To be clear, this works. Works fine afaict, actually, but I don't think I'll be able to get it into the main repo like this. If you're opposed to fixing, would you mind if I send a PR to include a nix install script (a nix flake) in your repo rather than the main Nix repo?

jurkovic-nikola, 2025-01-23:
Hello, is there any reason why you can't use /opt directory ? /opt should be used for this type of software, since its optional software you install if you're using Corsair hardware.

Tyler-Hardin, 2025-01-23:
Nix is another type of immutable OS, sorta. It's both immutable and "pure."
Packages can't create files outside of the chroot they build in,
effectively. There's also not really a concept of installing to /opt like
we would do on regular distros.

I don't really know Go, but I looked at the code a bit and I can see why
you wouldn't wanna do this. It would be a lot of lines changed.

What do you think about having it download a tarball (from github) to
initialize its working directory if it starts and finds static/ and/or web/
missing?

jurkovic-nikola, 2025-01-23:
Hello, please join https://discord.gg/FRhUGJRb to discuss this.

closed 2025-01-24 by jurkovic-nikola

skbolton, 2025-03-30:
Was there any resolution to this? I am also trying to get this up and running in nixos - the nix way.

Tyler-Hardin, 2025-04-09:
```
{ config, lib, pkgs, nixpkgs-unstable, ... }:
  let cfg = config.openlinkhub;
      pkg = cfg.package.overrideAttrs (oldAttrs: {
        postInstall = ''
          mkdir -p $out/share/OpenLinkHub
          cp -r database $out/share/OpenLinkHub
          cp -r static $out/share/OpenLinkHub
          cp -r web $out/share/OpenLinkHub

          mkdir -p $out/lib/udev/rules.d/
          cp *.rules $out/lib/udev/rules.d/
        '';
      });
  in {

  options = {
    openlinkhub.enable = lib.mkEnableOption "OpenLinkHub";
    openlinkhub.package = lib.mkPackageOption nixpkgs-unstable "openlinkhub" { };
  };

  config = lib.mkIf config.openlinkhub.enable {
    users.users.openlinkhub = { isSystemUser = true; group = "openlinkhub"; };
    users.groups.openlinkhub = {};

    systemd.services.OpenLinkHub = {
      description = "Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs";
      after = [ "sleep.target" ];
      wantedBy = [ "multi-user.target" ];
      script = ''
        for dir in database static web; do
          if ! [[ -d $dir ]]; then
            cp -r ${pkg}/share/OpenLinkHub/$dir .
            chmod -R u+w $dir
          fi
        done
        ${lib.getExe pkg}
        '';
      reload = "kill -s HUB $MAINPID";
      path = [ pkgs.pciutils ];

      serviceConfig = {
        User = "openlinkhub";
        Group = "openlinkhub";
        RestartSec = 5;
        StateDirectory = "OpenLinkHub";
        WorkingDirectory = "/var/lib/OpenLinkHub";
        ReadWritePaths = [ "/var/lib/OpenLinkHub" ];
      };
    };

    environment.systemPackages = [ pkg ];

    services.udev.packages = [ pkg ];
  };
}
```
I ended up getting rid of the overlay mount. This seemed cleaner to me. Haven't bothered to try to upstream it. I have a few PRs outstanding there already.