jurkovic-nikola/OpenLinkHub #65: Weird fragments on LCD text

Nytem, 2025-01-25:
I just updated to V0.4.9, but this has always been occurring on previous versions, I probably should've reported it a while ago even though it doesn't really bother me...

The strange part is that the fragments are always in the same spot for each character, which is why 0, 2, or 3 for example always appear fine. Not exactly sure if this is my LCD in specific, or if there's some issue with the font on it? I'm completely clueless.

[image]

Thanks for your work and time!

Distro: Fedora 41 (Used to be using PopOS where this issue still existed.)
Device: [iCUE H150i ELITE LCD XT](https://www.corsair.com/ca/en/p/cpu-coolers/cw-9060075-ww/icue-h150i-elite-lcd-xt-display-liquid-cpu-cooler), plugs in via the Commander Core.

jurkovic-nikola, 2025-01-25:
Hey, this has nothing to do with your LCD or USB connections. Its same on all 3 LCD types and it's caused by not using right font face on render. It will be fixed in 0.5.0 release, or if you compile code on your own, in next commit.

Nytem, 2025-01-26:
oh alrighty, thank you!

closed 2025-01-26 by Nytem