jurkovic-nikola/OpenLinkHub #68: Profile linear interpolation

LJspice, 2025-02-09:
**Describe the solution you'd like**
Add linear interpolation for target fan speed/pump speed based on temperature

**Additional context**
Step changes in fan speed are distracting, and result in unnecessary temperature cycling

jurkovic-nikola, 2025-02-10:
Hello, what's the issue with current Liquid profile template ? You can tweak fan / pump speed how you link based on liquid temperature, same goes for other profiles.

LJspice, 2025-02-12:
Understood that the target speed can be edited according to temperature

This feature request is for linear interpolation between those breakpoints

For example:
between 70-80C, fan speed is 70%
between 80-90C, fan speed is 90%

The current behavior is that the fan speed jumps from 70 to 90 when crossing 79.9 to 80.1 degrees

With linear interpolation, the target speed would ramp smoothly between breakpoints. At an example 78.6C, target fan speed would be 87.2%.

It looks like go has a useful package to implement this: [https://pkg.go.dev/gonum.org/v1/gonum/interp#PiecewiseLinear](https://pkg.go.dev/gonum.org/v1/gonum/interp#PiecewiseLinear)

jurkovic-nikola, 2025-02-15:
But how will linear interpolation solve this ? If you base your profile on CPU temperature instead of liquid, what happens if you jump from 40C to 70C with heavy workload ? Interpolation for this would require 5 inputs, current temp, min / max temp for given ID, min / max fan or pump speed. If you jump from 40 to 80C, you're gonna skip 3 ranges and interpolation will just ramp up fans same as profile does due to temperature difference even if only crossing points are used.

jurkovic-nikola, 2025-05-03:
New temperature profile system will be released in 0.5.6. This will include graph based profiles and interpolation.

closed 2025-05-18 by jurkovic-nikola