jurkovic-nikola/OpenLinkHub #69: missing Keys in K10 RGB Config

cnschu, 2025-02-09:
**Describe the bug**
in the config dialog of the Corsair K100 RGB Keyboard are two Keys missing ("|" and "#" on the German layout)
See photos and screenshot.

**To Reproduce**
1. Update from OpenLinkHub  0.3.3 to0.5.0
2. open http://127.0.0.1:27003/ in browser
3. click on "K100 RGB"
4. the above mentioned keys doesn't show up

**Expected behavior**
All keys of the K100 RGB should be visible an configurable in the interface

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint 22 Cinnamon
 - Device Firmware:  0.32.262
 - Program Version: 0.5.0

PabloGS, 2025-02-11:
I can second that for the K100 AIR RGB Firmware: 5.13.81, same issue:
<|> and #' keys are not illuminated. All other keys are fine.

jurkovic-nikola, 2025-02-11:
Yes, because DE layout was never made. Only US and second EU for template.

cnschu, 2025-02-14:
i assume the `packetIndex ` is the ID of the LED. How can i get this ID? i couldn't find any documentation containing this ID.

jurkovic-nikola, 2025-02-14:
There is no documentation about it, its just going each 3 indexes and seeing what LED turns on.

cnschu, 2025-02-18:
is there any way i can help?

cnschu, 2025-02-27:
[USB-Capure-#.pcapng.gz](https://github.com/user-attachments/files/19008663/USB-Capure-.pcapng.gz)
[USB-Capure-Bottom-Left.pcapng.gz](https://github.com/user-attachments/files/19008662/USB-Capure-Bottom-Left.pcapng.gz)

i hope it's of use to you.

country1706, 2025-02-27:
You can illuminate them, when you first illuminate the whole row and then change the colour of the other keys.
I also have a german layout and it worked for me this way, though its a bit of work

jurkovic-nikola, 2025-02-27:
> [USB-Capure-#.pcapng.gz](https://github.com/user-attachments/files/19008663/USB-Capure-.pcapng.gz) [USB-Capure-Bottom-Left.pcapng.gz](https://github.com/user-attachments/files/19008662/USB-Capure-Bottom-Left.pcapng.gz)
> 
> i hope it's of use to you.

Not much. Can you do another dump with light being disabled on all keys and only `#` and `< | >` are active.

cnschu, 2025-03-01:
The capures above are of a icue session, in wich the keys in question are turned off and on again mutiple times.
What exactly do you want captured if not the on off commands to the LEDs?

cnschu, 2025-03-01:
> Other thing you can do is to download https://github.com/jurkovic-nikola/OpenLinkHub/blob/main/database/keyboard/k100-de.json and put in `/opt/OpenLinkHub/database/keyboard/`. Restart service after that, change layout to DE on Web and restart service once again. This file contains packet index from K100 Air version, maybe it will work with K100.

hmmm.. it soesn't seem to be a german layout at all.
but i'll test some other configs.

cnschu, 2025-03-01:
> Other thing you can do is to download https://github.com/jurkovic-nikola/OpenLinkHub/blob/main/database/keyboard/k100-de.json and put in `/opt/OpenLinkHub/database/keyboard/`. Restart service after [...]

this config works fine.
thanks.

cnschu, 2025-03-02:
fixed the key remaining key labels and layout issues.

jurkovic-nikola, 2025-03-02:
Merged, thank you for PR and testing.

closed 2025-03-02 by jurkovic-nikola