jurkovic-nikola/OpenLinkHub #70: Corsair Nexus support

cnschu, 2025-02-09:
**Describe the solution you'd like**
It would be very nice to have the corsair nexus supported by OpenLinkHub.

CrimsonKnight13, 2025-03-27:
Additional info from lsusb

Device: ID 1b1c:1b8e Corsair CORSAIR iCUE NEXUS
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
bLength                18
bDescriptorType         1
bcdUSB               2.00
bDeviceClass            0 [unknown]
bDeviceSubClass         0 [unknown]
bDeviceProtocol         0
bMaxPacketSize0        64
idVendor           0x1b1c Corsair
idProduct          0x1b8e CORSAIR iCUE NEXUS
bcdDevice            2.00
iManufacturer           1 iDisplay Technology Co., Ltd.
iProduct                2 CORSAIR iCUE NEXUS
iSerial                 <removed>
bNumConfigurations      1
Configuration Descriptor:
bLength                 9
bDescriptorType         2
wTotalLength       0x0042
bNumInterfaces          2
bConfigurationValue     1
iConfiguration          0
bmAttributes         0xa0
(Bus Powered)
Remote Wakeup
MaxPower              288mA
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        0
bAlternateSetting       0
bNumEndpoints           2
bInterfaceClass         3 Human Interface Device
bInterfaceSubClass      0 [unknown]
bInterfaceProtocol      0
iInterface              0
HID Device Descriptor:
bLength                 9
bDescriptorType        33
bcdHID               1.10
bCountryCode            0 Not supported
bNumDescriptors         1
bDescriptorType        34 (null)
wDescriptorLength     245
Report Descriptors:
** UNAVAILABLE **
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x81  EP 1 IN
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0200  1x 512 bytes
bInterval               1
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x02  EP 2 OUT
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0400  1x 1024 bytes
bInterval               1
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        1
bAlternateSetting       0
bNumEndpoints           1
bInterfaceClass         3 Human Interface Device
bInterfaceSubClass      1 Boot Interface Subclass
bInterfaceProtocol      1 Keyboard
iInterface              0
HID Device Descriptor:
bLength                 9
bDescriptorType        33
bcdHID               1.10
bCountryCode            0 Not supported
bNumDescriptors         1
bDescriptorType        34 (null)
wDescriptorLength      45
Report Descriptors:
** UNAVAILABLE **
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x83  EP 3 IN
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0040  1x 64 bytes
bInterval              16
Device Qualifier (for other device speed):
bLength                10
bDescriptorType         6
bcdUSB               2.00
bDeviceClass            0 [unknown]
bDeviceSubClass         0 [unknown]
bDeviceProtocol         0
bMaxPacketSize0        64
bNumConfigurations      1
Device Status:     0x0000
(Bus Powered)

jurkovic-nikola, 2025-04-07:
Basic support added in 0.5.3. Options will / can expand based on community input.

closed 2025-04-08 by jurkovic-nikola