jurkovic-nikola/OpenLinkHub #74: Issue Starting OpenLinkHub

cklaubur, 2025-02-26:
I'm trying to run OpenLinkHub on Mint 22.1, and am having some trouble. The web GUI loads if I go to /opt/OpenLinkHub and run the script for it from there, but the system service fails to start, so none of my Corsair devices, a  Link Hub for some QX fans,  a Commander Core XT, a K60 RGB Pro keyboard and a Void Elite wireless headset.

The Systemctl status command gives me this wall of text:

casey@casey-System-Product-Name0://opt/OpenLinkHub$ sudo systemctl status OpenLinkHub.service
× OpenLinkHub.service - Open source interface for iCUE LINK System Hub, Corsair AIOs and Hubs
     Loaded: loaded (/etc/systemd/system/OpenLinkHub.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-02-25 20:47:48 MST; 2s ago
   Duration: 1.890s
    Process: 77408 ExecStart=/opt/OpenLinkHub/OpenLinkHub (code=exited, status=2)
   Main PID: 77408 (code=exited, status=2)
        CPU: 962ms

Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]: OpenLinkHub/src/devices/lsh.(*Device).getDevices(0xc000002480)
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]:         /home/nikola/Git/OpenLinkHub/src/devices/lsh/lsh.go:1829 +0x2110
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]: OpenLinkHub/src/devices/lsh.Init(0x1b1c, 0xc3f, {0xc00029caa0, 0x20?})
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]:         /home/nikola/Git/OpenLinkHub/src/devices/lsh/lsh.go:300 +0xf05
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]: OpenLinkHub/src/devices.Init.func2(0x0?, 0x0?, {0xc00029caa0?, 0x0?})
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]:         /home/nikola/Git/OpenLinkHub/src/devices/devices.go:1298 +0x1c
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]: created by OpenLinkHub/src/devices.Init in goroutine 1
Feb 25 20:47:48 casey-System-Product-Name0 OpenLinkHub[77408]:         /home/nikola/Git/OpenLinkHub/src/devices/devices.go:1297 +0xc7e
Feb 25 20:47:48 casey-System-Product-Name0 systemd[1]: OpenLinkHub.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Feb 25 20:47:48 casey-System-Product-Name0 systemd[1]: OpenLinkHub.service: Failed with result 'exit-code'.
casey@casey-System-Product-Name0://opt/OpenLinkHub$ 

I'm a bit stuck on how to proceed any further with this. I'm somewhat comfortable with  computers in general, but a bit of a novice to the land of Linux. Looks like the issue involves the "INVALIDARGUMENT" line, but I have no  clue what arguments are being passed that it's having trouble with. 

I tried checking stdout.log, but it's just a blank file. Is there a step I need to perform to get a log?

jurkovic-nikola, 2025-02-26:
If stdout.log is empty, this is probably permission issue. Check if openlinkhub username exists and check if this username is owner of /opt/OpenLinkHub directory. You should have at least 4 log entries before any device init happens, and logs are related to dashboard settings. Include also on what version you are.

cklaubur, 2025-02-26:
> If stdout.log is empty, this is probably permission issue. Check if openlinkhub username exists and check if this username is owner of /opt/OpenLinkHub directory. You should have at least 4 log [...]

Just did a full uninstall and reinstall using 0.5.1 to see if that fixed anything. The "openlinkhhub" username exists and is the owner of /opt/OpenLinkHub/.  The reinstall seems to have the logfile working now, but the service status still shows the same error.

Here's the log from the last attempt to start OpenLinkHub:
[stdout.log](https://github.com/user-attachments/files/18977787/stdout.log)

jurkovic-nikola, 2025-02-26:
Set debug value in config.json to true and start service. Send stdout.log with debug enabled. It crashes on device enumeration on Link System Hub and debug will print packet data. Please write what devices are connected to link system hub.

jurkovic-nikola, 2025-02-26:
Yeah, packet is fine, just processing of the same isn't. Probably will be fixed today.

jurkovic-nikola, 2025-02-26:
Issue is caused by not properly reading second packet when devices are enumerated. This only triggers when amount of devices passed 13. Its fixed and will be released in 0.5.2 version.

cklaubur, 2025-02-27:
> Issue is caused by not properly reading second packet when devices are enumerated. This only triggers when amount of devices passed 13. Its fixed and will be released in 0.5.2 version.

Just reinstalled from source, and it appears to work now. I can access the Dashboard and control everything. Well every supported thing at least. Thanks again for working on this.

closed 2025-02-27 by cklaubur