jurkovic-nikola/OpenLinkHub #75: Some problems detected.

Niil78, 2025-02-27:
Hi, I am an ArchLinux user. I am experiencing some errors with the following devices:

Corsair K70 KTL

Corsair Virtuoso MAX (Not detected)

Corsair H100i

K70KTL:

I am experiencing problems when configuring 8000Mhz.

The keyboard freezes quite frequently.

Also the software completely changes the behavior of the keyboard. The extra keys are no longer useful. 

Virtuoso MAX:

Not detected. I would like to know the battery level of this device.

H100i:

No values are shown in the dashboard.

Best regards. Great work

[image]

Niil78, 2025-02-28:
➜  ~ lsusb -d 1b1c:
Bus 002 Device 006: ID 1b1c:0c18 Corsair H100i Platinum
Bus 002 Device 008: ID 1b1c:2a02 Corsair CORSAIR VIRTUOSO MAX WIRELESS Gaming Receiver
Bus 005 Device 004: ID 1b1c:2b14 Corsair CORSAIR K70 PRO RGB TKL Magnetic Mechanical Gaming Keyboard

The problem with the keyboard is that it freezes and sometimes malfunctions. every so often it is like it gets stuck, and if it gets stuck while you are pressing a key it stays pressed for 1 or 2 seconds. It only happens to me if I set the keyboard to 8000.
If I set it lower it doesn't give me any problems. 

And yes you're right, I didn't have the show all devices thing.

jurkovic-nikola, 2025-03-25:
Support for Virtuoso MAX is added in latest commit. Regarding K70, I just can't make it to break on my system. Tried multiple physical machines and VMs.