jurkovic-nikola/OpenLinkHub #76: K100 AIR - Function Keys (Slipstream or Bluetooth)

zer0signa1, 2025-02-27:
Hello,  I have been using OpenLinkHub since Nov. 2024.  I love it, there a few quirks, but nothing game breaking.  Thank you so much for this!!

I am looking to see if there is a way to implement the device switch that's built into the K100 AIR.  With it being wireless you have the option to use USB, Slipstream (wireless), or Bluetooth.   This is achieved by pressing the FN Key +  G1 (Slipstream), or G2 (Bluetooth).

Currently when the device is held by OpenLinkHub, that feature does not work.  They keyboard doesn't even recongize the key combo via the connection status light.  The use case is Work PC or Personal PC.  

If this can't be implemented, what's the easiest way to blacklist the K100 from OpenLinkHub.

I tried comment out the UDEV Rule that's linked to this USB Device "1ba6"  (in /etc/udev/rules.d/99-Openlinkhub, as well as the /opt/Openlinkhub location that;  But if I do this OpenLinkHub Servic fails to load.

Thanks, keep up the great work!

```
lsusb |grep -i cors
Bus 003 Device 017: ID 1b1c:1c1f Corsair HX1500i Power Supply
Bus 003 Device 018: ID 1b1c:0c10 Corsair Commander PRO
Bus 003 Device 020: ID 1b1c:0c3f Corsair iCUE LINK System Hub
Bus 007 Device 003: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver

```

jurkovic-nikola, 2025-02-28:
Hello,

There is a way to implement this. I'll push it out in 0.5.2 for default key actions. Regarding blocking the device from software, you have in config.json  `"exclude": []` field that represent uint16 value of PID. In your case, value will be `"exclude": [7078]`. This is remove complete slipstream from being processed by the program. You don't need to modify any udev rules.

zer0signa1, 2025-02-28:
I appreciate the effort for pushing this update in the future!   

The exclude: function was exactly what I was looking for, for the time being!  It has worked, and I can continue to use the other features of your product.

I forgot to mention the other thing I was thinking about w/ this program and that is the API.  Currently I use the Stream Controller Software to display stats on my hardware most if it is just curl/sys commands.  

I was looking through the API Documentation and was hoping there was an API Function for being able to query the Temperature that the Corsiar ICUE Link QX120 Fans report.  I didn't see a function for it; I was able to crudely do this with Curl and a few greps and awks.

Would this be something easy to implement into the software?

```
┌─[:)]─[07:11:26]─[signal@265k]─[~]
└──> cat Documents/Automation/stream-controller/water-temp.sh 
#!/usr/bin/env bash
curl -X GET http:/127.0.0.1:27003/api/ --silent |jq |grep -A5 01000CF6420356179100014EF9 |head -6 |grep temperatureString |awk -F '"' '{print $4}'

┌─[:)]─[07:11:54]─[signal@265k]─[~]
└──> curl -X GET http:/127.0.0.1:27003/api/ --silent |jq |grep -A5 01000CF6420356179100014EF9 |head -6 |grep temperatureString |awk -F '"' '{print $4}'
25.1 °C
```

_Do you also have some kind of BuyMe a Coffee/paypal link?  I would love to donate to you this project?_

jurkovic-nikola, 2025-03-01:
There is no custom function for Link System Hub. You can query device directly if you know serial number, it shouldn't be hard to pull it via `lsusb` then pass it to `http://127.0.0.1:27003/api/devices/40027074EFEBF2568288ACE590128B30`. This will return complete hub data. Python would be better approach to working with json instead of bash script. With python you can make a data model and parse data like that. 

`"HasSpeed": true` - Device has RPM value. 
`"HasTemps": true` - Device has `temperatureString` and `temperature` value. 
`"AIO": true` - Device is AIO. 

There is no donation for this.

zer0signa1, 2025-03-23:
Thank you so much for the Device Switching functionality.  It works great in 5.2!

I will close this issue!

closed 2025-03-23 by zer0signa1