jurkovic-nikola/OpenLinkHub #81: H115i RGB Platinum not showing since upgrade

Nortfor, 2025-03-02:
**Describe the bug**
Ever since the upgrade via git clone, the device has disappeared from the dashboard.

**To Reproduce**
Steps to reproduce the behavior:
1. Your configuration values
2. Number of devices
3. Content of `/opt/OpenLinkHub/stdout.log`
`{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is missing, creating initial one.","time":"2025-03-02T16:15:58+01:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is created.","time":"2025-03-02T16:15:58+01:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is missing, creating initial one.","time":"2025-03-02T16:15:58+01:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is created.","time":"2025-03-02T16:15:58+01:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"080500DC040E0712","time":"2025-03-02T16:15:59+01:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"3095","time":"2025-03-02T16:15:59+01:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"080500DC040E0712","time":"2025-03-02T16:15:59+01:00"}
{"error":"listen tcp 127.0.0.1:27003: bind: address already in use","level":"fatal","msg":"Unable to start REST server","time":"2025-03-02T16:15:59+01:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2025-03-02T16:22:08+01:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2025-03-02T16:22:08+01:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Scheduler file is found, checking for any upgrade...","time":"2025-03-02T16:22:08+01:00"}
{"file":"/opt/OpenLinkHub/database/scheduler.json","level":"info","msg":"Nothing to upgrade.","time":"2025-03-02T16:22:08+01:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"3095","time":"2025-03-02T16:22:09+01:00"}
{"level":"warning","msg":"No profile found for device. Probably initial start","serial":"080500DC040E0712","time":"2025-03-02T16:22:09+01:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"080500DC040E0712","time":"2025-03-02T16:22:09+01:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"080500DC040E0712","time":"2025-03-02T16:22:09+01:00"}
{"level":"warning","msg":"DeviceProfile is not set, probably first startup","serial":"080500DC040E0712","time":"2025-03-02T16:22:09+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:10+01:00"}
{"level":"info","msg":"No RGB compatible devices found","time":"2025-03-02T16:22:10+01:00"}
{"level":"info","msg":"Device successfully initialized","product":"Commander PRO","serial":"080500DC040E0712","time":"2025-03-02T16:22:10+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:15+01:00"}
{"level":"info","msg":"No RGB compatible devices found","time":"2025-03-02T16:22:15+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:20+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:25+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:30+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:34+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:38+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:42+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:46+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:22:53+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:04+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:08+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:11+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:13+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:20+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:32+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:38+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:44+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:23:55+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:04+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:11+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:15+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:18+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:20+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:22+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:24:30+01:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Dashboard file is found, checking for any upgrade...","time":"2025-03-02T16:24:53+01:00"}
{"file":"/opt/OpenLinkHub/dashboard.json","level":"info","msg":"Nothing to upgrade.","time":"2025-03-02T16:24:53+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:25:01+01:00"}
{"level":"info","location":"/opt/OpenLinkHub/database/profiles/080500DC040E0712.json","msg":"Loaded custom user profile","serial":"080500DC040E0712","time":"2025-03-02T16:25:06+01:00"}
[... cut ...]
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

**Expected behavior**
Device to be showing on the dashboard.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: CachyOS
 - Program Version [Build: cae9e31ddbb636e3de807d8ee0e6870b7845b479 / 0.0.0]

**Additional context**
I'm still very new to Linux, so apologies if there are things I don't know yet, or aren't helpful
Also could the K70 RGB PRO Optical-Mechanical Gaming Keyboard be added? (Bus 001 Device 005: ID: 1b1c:1bd4)

Thanks in advance!
~ Nort

jurkovic-nikola, 2025-03-02:
Try to follow Uninstall section from readme file, reboot and install again. There was no change for this device in recent commits.

Nortfor, 2025-03-02:
After the complete uninstall, reboot and installing it again, it still does not show up within the dashboard, only the Commander Pro does

Nortfor, 2025-03-02:
After a reboot into Windows afterwards (constant rebooting back into CachyOS made no difference) and then rebooting back into Linux again the device has showed up again within the Dashboard, no idea what might've been causing that with the AIO but at least it has been resolved!

On another note however, would you like me to open a feature request for the K70 keyboard or leave this one open?

Thanks for the help!

jurkovic-nikola, 2025-03-02:
Please open feature request for keyboard. If device can be obtained, it will be implemented.

Nortfor, 2025-03-02:
Made the feature request, will be closing this one! Thanks again!

closed 2025-03-02 by Nortfor