jurkovic-nikola/OpenLinkHub #85: UI showing error : non existing LED port-id commander core

IOTech17, 2025-03-05:
**Describe the bug**
When trying to add the LED from port 6, the UI show an error : Non Existing LED Port-ID

**To Reproduce**
Steps to reproduce the behavior:
1. Your configuration values
2. 6 fans + 6 LED connected to the commander core

**Desktop (please complete the following information):**
 - OS: Solus OS OpenHubLink was installed manually
 - firmware : 2.0.19
 - Program Version : 0.5.4

**Additional context**
Add any other context about the problem here.

IOTech17, 2025-03-06:
@jurkovic-nikola 
I have no error in the logs,

I can see the led flashing during boot time on the specific fan connected on port RGB 6
Bus 001 Device 004: ID 1b1c:0c32 Corsair CORSAIR iCUE COMMANDER Core

This is the profile.json file I have.

I am trying to force it via the json file but the LED is not recognized, however it does work upon boot and on windows.

Any idea why?

`{
    "Active": true,
    "Path": "/opt/OpenLinkHub/database/profiles/2071357B3334.json",
    "Product": "iCUE COMMANDER Core",
    "Serial": "2071357B3334",
    "LCDMode": 0,
    "LCDImage": "",
    "LCDRotation": 0,
    "Brightness": 0,
    "BrightnessSlider": 100,
    "OriginalBrightness": 100,
    "RGBProfiles": {
        "0": "cpu-temperature",
        "1": "cpu-temperature",
        "2": "cpu-temperature",
        "3": "cpu-temperature",
        "4": "cpu-temperature",
        "5": "cpu-temperature"
    },
    "SpeedProfiles": {
        "0": "Normal",
        "3": "Liquid",
        "4": "Liquid",
        "5": "Liquid",
        "6": "Liquid"
    },
    "Labels": {
        "0": "Set Label",
        "3": "Set Label",
        "4": "Set Label",
        "5": "Set Label",
        "6": "Set Label"
    },
    "RGBLabels": {
        "0": "Set Label",
        "1": "Set Label",
        "2": "Set Label",
        "3": "Set Label",
        "4": "Set Label",
        "5": "test"
    },
    "CustomLEDs": {
        "1": 6,
        "2": 6,
        "3": 0,
        "4": 0,
        "5": 0,
        "6": 0
    }
}`

jurkovic-nikola, 2025-03-07:
Fans should be automatically detected when program starts. Manual section for ARGB is only for non-corsair devices. Flip debug to true in config.json and restart service. Send stdout.log after that. What fans do you have ?

IOTech17, 2025-03-21:
@jurkovic-nikola 

I fixed the issue by turning off the pc, then switching the LED connector between the coolermaster fan and the corsair fan and now all the port are working

closed 2025-03-21 by jurkovic-nikola