jurkovic-nikola/OpenLinkHub #87: k70 core rgb: Unable to change device RGB profile

kokoko3k, 2025-03-06:
**Describe the bug**
I'd like to add a custom rgb user profile and load it from command line.

**To Reproduce**
open K70 core RGB settings
select keyboard profile
edit it
save as "myprofile"
curl -X POST http:/127.0.0.1:27003/api/color -d '{"deviceId":"05025016AFDC9027656AA670F5001BC3",  "profile":"myprofile"}' --silent  | jq

```{
  "code": 200,
  "status": 0,
  "message": "Unable to change device RGB profile"
}```

The new profile can still be called via the  "keyboard" profile instead.

**Expected behavior**
To load my named saved profile

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Device Firmware: Firmware: 1.21.72
 - Program Version 0.5.1

jurkovic-nikola, 2025-03-06:
You neee to save user profile, and use /api/userProfile method to switch profiles.

jurkovic-nikola, 2025-03-06:
API docs are updated from time to time, i'll take a look later on.

thomAguirre, 2025-03-07:
I was having the same issue. Looking at "/src/server/requests/requests.go", seems like the correct json tag is "userProfileName"

`
curl -X POST http:/127.0.0.1:27003/api/userProfile -d '{"deviceId":"05025016AFDC9027656AA670F5001BC3", "userProfileName":"myprofile"}' --silent | jq
`

Above worked for me.

kokoko3k, 2025-03-07:
Thanks,
but that way only "default" profile works, for anything else, where the webui is able to switch between default, koko and koko2 profiles, i always get:
```
{
  "code": 200,
  "status": 0,
  "message": "Unable to change user profile"
}
```

kokoko3k, 2025-03-07:
Ok, in inspected the webui from filrefox and found this that works:

``` 
curl -X POST http:/127.0.0.1:27003/api/keyboard/profile/change -d '{"deviceId":"05025016AFDC9027656AA670F5001BC3", "keyboardProfileName":"koko"}' --silent | jq
```

jurkovic-nikola, 2025-03-07:
There are 2 types of profiles for keyboard stuff, one is user profile and its generic for every device, second one is keyboard profile on top of generic profile. Docs are updated from time to time, but yeah, everything can be found in static js files and browser inspect window.

kokoko3k, 2025-03-07:
Yeah, I noticed that, unfortunately the user profile does not change the
colors for some reason.

closed 2025-03-07 by kokoko3k